matthewwall/weewx-interceptor #1: Don't log missing data for WS1400

claudobahn, 2016-05-13:
I've got a WS1400 w/ ObserverIP with firmware 3.0.7. Right now I've only got the indoor sensor hooked up, here's what an update looks like:

`/weatherstation/updateweatherstation.php?ID=xxx&PASSWORD=xxx&tempf=-9999&humidity=-9999&dewptf=-9999&windchillf=-9999&winddir=-9999&windspeedmph=-9999&windgustmph=-9999&rainin=0.00&dailyrainin=0.00&weeklyrainin=0.00&monthlyrainin=0.00&yearlyrainin=0.00&solarradiation=-9999&UV=-9999&indoortempf=66.2&indoorhumidity=47&baromin=29.94&lowbatt=0&dateutc=2016-5-10%202:34:15&softwaretype=Weather%20logger%20V3.0.7&action=updateraw&realtime=1&rtfreq=5`

As you can see, the missing sensor values are reported as `-9999` with the exception of rain (probably a bug on their end). Currently, most of those `-9999` values make it into the weewx database. I think it probably would make sense to add handling to filter those out?

Happy to break these out into separate issues if you prefer, but I didn't want to spam you:
- I'm able to telnet into my ObserverIP (l: admin, p: admin) and change the host it reports to using `setdsthn some.new.host` followed by `saveconfig` which then has my ObserverIP reporting to weewx without any fancy setup.
- With the above configuration, a reverse proxy seems most useful, because it allows me to bind the weewx interceptor to 127.0.0.1 and serve up weewx on public port 80, example Apache configuration:

>         ProxyPass         "/weatherstation/updateweatherstation.php" "http://127.0.0.1:5971/weatherstation/updateweatherstation.php"
        ProxyPassReverse  "/weatherstation/updateweatherstation.php" "http://127.0.0.1:5971/weatherstation/updateweatherstation.php"

- Would be nice to enforce a station & password for accepting reports (very low priority)
- Would be incredibly useful to have a setting in the interceptor that would turn on a pass-through to `rtupdate.wunderground.com`, perhaps after having eliminated `-9999` measurements.

matthewwall, 2016-05-13:
> As you can see, the missing sensor values are reported as -9999 with the exception of rain (probably a bug on their end). Currently, most of those -9999 values make it into the weewx database. I think it probably would make sense to add handling to filter those out?

fixed at commit 9d84aeb

matthewwall, 2016-05-13:
> Would be nice to enforce a station & password for accepting reports (very low priority)

not sure what you mean.  do you mean to add name/pass fields to the weewx.conf config, then have the driver __init__ do the telnet and setdsthn/saveconfig?

matthewwall, 2016-05-13:
> Would be incredibly useful to have a setting in the interceptor that would turn on a pass-through to rtupdate.wunderground.com, perhaps after having eliminated -9999 measurements.

i started doing this some time ago, then i realized that weewx.restx.StdWunderground already does the upload to weather underground.

claudobahn, 2016-05-13:
@matthewwall thanks for the incredibly quick response & fix!

> not sure what you mean. do you mean to add name/pass fields to the weewx.conf config, then have the driver init do the telnet and setdsthn/saveconfig?

Sorry for not being clear. My thought was that the interceptor could be configured for a particular station id and then match the incoming station packets to make sure only packets from a known station are accepted. Admittedly, it's a fairly unlikely scenario that somebody is going to send fake data.

> i started doing this some time ago, then i realized that weewx.restx.StdWunderground already does the upload to weather underground.

Do you know by any chance whether any data transformation happens? I guess I'll just try it out, thanks for the tip!

matthewwall, 2016-05-14:
could you send a complete capture of the telnet session for setdsthn?

claudobahn, 2016-05-14:
Certainly, here you go. Simply replace `rtupdate.wunderground.com` with the hostname of your choosing.

> telnet 172.16.1.13
> Trying 172.16.1.13...
> Connected to 172.16.1.13.
> Escape character is '^]'.
> username: admin
> password: 
> Successful login through telnet
> telnet> setdsthn rtupdate.wunderground.com
> Ok
> telnet> saveconfig
> Saving Configuration to FLASH
> Ok
> telnet> rebootConnection closed by foreign host.
> 

To restore it back to wunderground, the above snippet will do nicely.

claudobahn, 2016-05-14:
Looks like the password didn't echo back, didn't notice, sorry. It's `admin`, as well.

claudobahn, 2016-05-30:
fyi, I've been using this with great success for a week now. My setup is:
- Ambient Weather WS1400IP (equivalent to Fine Offset WH2600 I believe) w/ both indoor and outdoor sensing
- [Firmware version 3.0.7](http://site.ambientweatherstore.com/iptools/USA_v3.0.7.bin)
- Apache w/ reverse proxy to the interceptor bound to 127.0.0.1 as described above

Everything seems to work well, including the missing data handling, so I'm closing this issue.

closed 2016-05-30 by claudobahn