matthewwall/weewx-interceptor #3: lacross bridge

Jayhawk85, 2016-06-13:
When I try to use the lacross-bridge switch, the driver install fails with unsupported driver type

matthewwall, 2016-06-13:
in weewx.conf, change to 'lacrosse-bridge' instead of 'lacross-bridge':

device_type = lacrosse-bridge

i must warn you, the driver will probabaly fail even after that.  i do not have lacrosse hardware on which to test.  would you be willing to provide me with remote access to the computer running weewx so i can get everything to work?

Jayhawk85, 2016-06-13:
actually, just tried that and it loaded (at least according to syslog)  Just a typo in the notes

Jayhawk85, 2016-06-13:
Sure (trying to figure out how we would do that).

matthewwall, 2016-06-13:
- make a user account for mwall:

sudo adduser mwall

- be sure sshd is installed and running

- make port 22 accessible to the internet by mapping some port on your router's WAN interface to port 22 on the computer running weewx

Jayhawk85, 2016-06-13:
open.  give me an email to send the pw to  ip is 67.217.9.126

matthewwall, 2016-06-13:
mwall@users.sourceforge.net

Jayhawk85, 2016-06-13:
on the way

Jayhawk85, 2016-06-13:
were you able to get in?

matthewwall, 2016-06-13:
continuing the conversation in email

matthewwall, 2016-06-13:
as of commit e4a4b74 the driver is working with lacrosse GW1000U for the current observations

closed 2016-06-13 by matthewwall