matthewwall/weewx-interceptor #4: Help Loading Interceptor for use with Lacrosse Bridge

KC1ELF, 2016-08-05:
I am no programmer and new to Linux.  I installed the weewx program which is successfully outputting "simulator" data to PWSweather.

I get the following when I try to load the Interceptor........anyone know what I might be doing wrong?

tom@weatherham:~$ wee_extension --install weewx-interceptor.zip
Request to install 'weewx-interceptor.zip'
Extracting from zip archive weewx-interceptor.zip
Traceback (most recent call last):
  File "/usr/bin/wee_extension", line 83, in <module>
    main()
  File "/usr/bin/wee_extension", line 75, in main
    ext.install_extension(options.install)
  File "/usr/share/weewx/weecfg/extension.py", line 107, in install_extension
    self.tmpdir, self.logger)
  File "/usr/share/weewx/weecfg/__init__.py", line 1212, in extract_zip
    zip_archive = zipfile.ZipFile(open(filename, mode='r'))
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file

matthewwall, 2016-08-05:
it looks like the weewx-interceptor.zip file is corrupt.  how did you download it?

KC1ELF, 2016-08-05:
Hi Matthew,

I downloaded it this way:

tom@weatherham:~$ wget -o weewx-interceptor.zip https://github.com/matthewwall/weewx-interceptor/archive/master.zip

It does show up at the bottom of the weewx.confg file, but I'm not sure if it is working properly.

Thank you

matthewwall, 2016-08-05:
the zip file is corrupt.  try this instead:

wget --no-check-certificate -O weewx-interceptor.zip https://github.com/matthewwall/weewx-interceptor/archive/master.zip

closed 2016-08-05 by matthewwall

KC1ELF, 2016-08-20:
Hello Matthew, 

 

  Thank you for creating the Interceptor driver.  I am running a Lacrosse C84612 with the GW 1000 U gateway.  For the last year, I have been successfully uploading to WU using SkySpy on a windows machine.  Because SpySky is limited to only communicating with WU, I decided to build a dedicated Linux machine to run Weewx/Interceptor on.   The problem I am having is that the software is not fetching the data from the GW 1000 U at regular intervals.   I can see the software pinging the gateway successfully, but it does not always retrieve the data packets.  I know (from monitoring the SkySpy software, and from visually watching the gateway lights) that the C84612 sends a data packet to the GW 1000U gateway every 12 minutes.  The SkySpy software picks up the data like clockwork every 12 minutes, but with the Weewx, it might pick up 3 packets every 12 minutes, but then won’t return anything for 25 or 38, or sometimes more than an hour.  There doesn’t seem to be a repeating pattern, it’s just fairly random.  

 

You can see an example of what I am describing here:

https://www.wunderground.com/personal-weather-station/dashboard?ID=KMAHAVER17#history/s20160818/e20160818/mdaily

 

The wind speed graph show the best depiction of the data points.  I was running the Weewx from 12AM until 8:18 PM, at which time I switched back to the windows machine and the SkySpy software.

 

I tried running Weewx on several different ports because I originally thought the port I started with (80) might have been occupied, but the results are the same no matter which port I use.

 

I have very little knowledge of programming software, so I am getting close to throwing in the towel on this project.  Reaching out to you is my last hope….lol.

 

Any ideas what could be going on?  Do you know of anyone who has been successful with my lacrosse hardware set-up?

 

Best Regards,

 

Tom Kavanaugh