matthewwall/weewx-interceptor #8: Database not updating - problems running driver directly

prplrhd, 2016-11-28:
Matthew - greatly appreciate your efforts here. 

Before the firmware update on the acurite bridge, I had 2 systems working flawlessly, pi with acurite 5n1 sensor + secondary sensor (temp, humidity) + acurite bridge and weewx using a modified version of the hackulink driver. Basically just tweaked that driver to integrate the sniffer feed since I have the acurite bridge connected through the pi (network bridge).

Have updated to weewx 3.6.2 and installed latest Interceptor driver 0.17j, pcap installed etc. Interceptor driver configures ok, weewx.conf updated for sniff parameters and debug=1. weewx starts and runs fine but never appears to capture anything as it's not updating the database. syslog is completely quiet once weewx starts up, database file is untouched. If I configure simulator, all is well, database updates, web files created, updates in syslog from weewx, etc

Hardware setup unchanged, if I run ngrep at command line I see traffic passing, and traffic is passing just fine to acurite site. Obviously would be ideal if I could run interceptor.py directly for debug, and you anticipated such a need in your readme. Found this: 

PYTHONPATH=bin python bin/user/interceptor.py --device=acurite-bridge --mode=sniff --iface=eth0 --pcap_filter="src 192.168.0.4 and dst port 80"

However as written does not work for me (updated of course for proper iface & filter values), First, i'm not pathed to bin/user/interceptor.py. If I run with the full path as /usr/share/weewx/user/interceptor.py, I receive error"ImportError: No module named weewx.drivers" line 162

Do I need an include somewhere in the path? Run from specific directory? Something else?

Thanks again!

Matt

matthewwall, 2016-11-28:
matt,

for a deb install you want this:

`PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/interceptor.py --help`
or

```
cd /usr/share/weewx
PYTHONPATH=. python user/interceptor.py --help
```

you'll want to run that in debug mode to start with - use the --debug option

before you try to sniff with the interceptor, you might want to verify the traffic you're trying to sniff.  try tcpdump directly:

`sudo tcpdump -Anpl -s0 -w - -i eth0 src 192.168.0.4 and dst port 80 | stdbuf -oL strings -n8`

to be sure the traffic is where you think it is.

prplrhd, 2016-11-28:
Thanks Matthew – the tcpdump tip revealed the issue. Learned something new in the process…

 

ngrep and the other commands I had used previously to capture and pipe traffic would pick up off either eth0 or br0. Had been using br0 for no specific reason, did not occur to me that tcpdump would behave differently however eth0 is the actual hardware interface point for the origination of the traffic.

 

tcpdump only captures the traffic on eth0. Once I pointed interceptor to eth0, all is well.

closed 2016-12-03 by matthewwall