matthewwall/weewx-interceptor #10: Observer response handling (listen mode)

hoopty, 2017-01-04:
First off, thanks for contributing this driver, it made using weewx straightforward an possible for me.  Look forward to it making it into the weewx distro when ready.

I would like the Observer "server" status light to properly represent if the interceptor did indeed receive the data, like it does with WU.  My understanding is the interceptor should simply respond with the string "success" (or "failure") when the Observer publishes the WU data to it.  Please have interceptor emulate the WU behaviour so the Observer will operate as expected.  To clarify - I consider it a success if the interceptor received and parsed the data sent, I do not expect it to block and send data to WU/etc. before responding to the Observer.

Recent Observer firmware (for Ambient at least) has a "System Reboot" feature (If server communication is lost for 20 minutes, the ObserverIP will automatically reboot.), and I do not want to have to disable that setting (or have my Observer reboot every 20 minutes) as it has helped in the past with other Observer issues/hiccups.

matthewwall, 2017-01-04:
this should be fixed at commit 68096b1 (driver version 0.24rc1)

please let me know whether it works for you

hoopty, 2017-01-04:
Took you long enough.  ;)

Works as expected, thanks!

closed 2017-01-04 by hoopty