matthewwall/weewx-interceptor #13: sniff mode issue on raspian jessie 2017-01-11

Ketonom, 2017-02-12:
I've installed weewx using setup.py and then installed the pcap dependencies using "sudo apt-get install python-libpcap".  Afterwards, I installed the interceptor.py driver and configured it for the Acurite smartHUB as described in the readme file.

I currently have both my Raspberry PI (192.168.1.92) and Acurite smartHUB (192.168.1.91) connected to an old Linksys hub I had laying around. However, packets were not getting captured and parsed after I started weewx.  I then ran the driver directly in debug mode, and still no packets were being captured.  The syslog would just repeat "interceptor: MainThread: empty queue" every 10 seconds.

**/var/log/syslog**
_Feb 12 01:07:23 kodama interceptor[4619]: interceptor: MainThread: mode is sniff
Feb 12 01:07:23 kodama interceptor[4619]: interceptor: MainThread: sniff iface eth0
Feb 12 01:07:23 kodama interceptor[4619]: interceptor: MainThread: sniff filter 'src 192.168.1.91 and dst port 80'
Feb 12 01:07:23 kodama interceptor[4619]: interceptor: ServerThread: start sniff server
Feb 12 01:07:33 kodama interceptor[4619]: interceptor: MainThread: empty queue
Feb 12 01:07:43 kodama interceptor[4619]: interceptor: MainThread: empty queue
Feb 12 01:07:53 kodama interceptor[4619]: interceptor: MainThread: empty queue_

After this, I decided to try running tcpdump with the same filter that I was using with the interceptor.py  and it worked just fine.  Packets were being dumped to my console session.

**/var/log/syslog**
_Feb 12 01:08:25 kodama kernel: [ 4217.605007] device eth0 entered promiscuous mode
Feb 12 01:08:27 kodama kernel: [ 4219.575931] device eth0 left promiscuous mode_

**console**
_01:08:26.317722 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [S], seq 3925768075, win 400, options [mss 536], length 0
01:08:26.386360 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [.], ack 2301849652, win 400, length 0
01:08:26.388198 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 0:80, ack 1, win 400, length 80: HTTP: GET /weatherstation/updateweatherstation?dateutc=now&action=updateraw&realtime=1[!http]
01:08:26.452530 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 80:121, ack 1, win 400, length 41: HTTP
01:08:26.455467 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 121:144, ack 1, win 400, length 23: HTTP
01:08:26.463229 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 144:180, ack 1, win 400, length 36: HTTP
01:08:26.464736 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 180:197, ack 1, win 400, length 17: HTTP
01:08:26.466621 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 197:238, ack 1, win 400, length 41: HTTP
01:08:26.519580 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [P.], seq 238:261, ack 1, win 400, length 23: HTTP
01:08:26.589732 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [.], ack 401, win 400, length 0
01:08:26.650494 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [.], ack 413, win 389, length 0
01:08:26.651566 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [.], ack 413, win 400, length 0
01:08:26.652844 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [F.], seq 261, ack 413, win 400, length 0
01:08:26.653490 IP 192.168.1.91.2288 > ec2-34-199-173-158.compute-1.amazonaws.com.http: Flags [R.], seq 261, ack 413, win 400, length 0
^C_

The difference I noticed in the syslog was that eth0 would enter promiscuous mode when I ran tcpdump but wouldn't when I ran the interceptor.py script by itself.  When I decided to run them both at the same time, the interceptor.py script was picking up the packets too.

**console**
_weewx@kodama:~ $ sudo PYTHONPATH=bin python bin/user/interceptor.py --device=acurite-bridge --mode=sniff --iface=eth0 --filter="src 192.168.1.91 and dst port 80"
identifiers: {'bridge_id': '--removed--', 'sensor_type': 'tower', 'sensor_id': '--removed--'}
identifiers: {'bridge_id': '--removed--', 'sensor_type': 'tower', 'sensor_id': '--removed--'}
identifiers: {'bridge_id': '--removed--', 'sensor_type': '5N1x38', 'sensor_id': '--removed--'}
identifiers: {'bridge_id': '--removed--', 'sensor_type': 'tower', 'sensor_id': '--removed--'}
identifiers: {'bridge_id': '--removed--', 'sensor_type': '5N1x31', 'sensor_id': '--removed--'}
identifiers: {'bridge_id': '--removed--', 'sensor_type': 'tower', 'sensor_id': '--removed--'}
identifiers: {'bridge_id': '--removed--', 'sensor_type': 'tower', 'sensor_id': '--removed--'}
_

Would you be able to take a look at why the script isn't triggering promiscuous mode on eth0 by itself?

matthewwall, 2017-02-12:
at line 297 of interceptor.py, change this:

    PROMISCUOUS = 0

to this:

    PROMISCUOUS = 1

then restart weewx.

Ketonom, 2017-02-12:
Ah! That has fixed it, of course.  I wish I had thought to search the source for the word "promiscuous" prior to now.  Thanks for the assistance!

closed 2017-02-12 by Ketonom

matthewwall, 2017-02-12:
promiscuous is a config file option as of driver version 0.27 at commit 22bd5186ed6d2bbf31eb7a6a6bdb7b55340f4fe2