matthewwall/weewx-interceptor #14: sniff mode problem

regis44, 2017-02-17:
Hello,

I have weewx and weewx interceptor installed on a raspberry pi 2. I have an OS lw301 plugged into an usb-ethernet dongle eth1 bridged with eth0. The weather station is able to upload data to oregon server.

When weewx interceptor is uninstalled, weewx works in simulation mode and refresh the reports every 5 minutes. When weewx is installed in sniff mode (even in listen mode), it seems that weewx is "blocked".

My configuration is :

```
[Interceptor]
    # This section is for the network traffic interceptor driver.

    # The driver to use:
    driver = user.interceptor

    # Specify the hardware device to capture.  Options include:
    #   acurite-bridge - acurite internet bridge
    #   observer - fine offset WH2600/HP1000/HP1003, aka 'observer'
    #   lw30x - oregon scientific LW301/LW302
    #   lacrosse-bridge - lacrosse GW1000U/C84612 internet bridge
    device_type = lw30x
    mode = sniff
    iface = eth1
    pcap_filter = src 192.168.0.119 and dst port 80
```
Every time the lw301 upload data, I've logged something like this : 
```
Feb 17 09:39:02 raspberrypi weewx[1561]: interceptor: ServerThread: sniff: timestamp=1487320742.59 pktlen=60 data=00 24 d4 b4 80 5e 00 04 a3 69 03 fe 08 00 45 00 00 28 02 6f 00 00 64 06 8e fd c0 a8 00 77 36 e4 cd 60 04 43 00 50 c6 cb 6b c4 da 62 0a 4d 50 10 01 00 cd 9d 00 00 00 00 00 00 00 00
Feb 17 09:39:02 raspberrypi weewx[1561]: interceptor: ServerThread: sniff: timestamp=1487320742.59 pktlen=60 data=00 24 d4 b4 80 5e 00 04 a3 69 03 fe 08 00 45 00 00 28 02 70 00 00 64 06 8e fc c0 a8 00 77 36 e4 cd 60 04 43 00 50 c6 cb 6b c4 da 62 0a 4d 50 11 01 00 cd 9c 00 00 00 00 00 00 00 00
Feb 17 09:39:09 raspberrypi weewx[1561]: interceptor: MainThread: empty queue
```
I've spent a lot of time and restart from the beginning several times without success. I need help ! Do you an advice ?

THX.

Régis.

matthewwall, 2017-02-17:
the sniffer is capturing packets, but it is not finding anything useful in the packets.

neither of the two packets you posted contain an http GET or POST.  is 192.168.0.119 the ip address of the lw301?

do all the packets look like the two you posted?

regis44, 2017-02-17:
The lw301's IP is 192.168.0.109 on eth1 (usb dongle) connected to the lw301. 
The pi is connected to my ISP box on eth0 (pi's native ethernet connector), she has IP 192.168.0.108. I can access to the apache server page (192.168.0.118/weewx) via an another computer.

I changed the sniffed IP in the interceptor configuration to 192.168.0.118 and now I have logs that contain POST request and station datas.

```
Feb 17 15:55:20 raspberrypi weewx[809]: interceptor: ServerThread: sniff: timestamp=1487343320.44 pktlen=191 data=00 24 d4 b4 80 5e 00 04 a3 69 03 fe 08 00 45 00 00 b1 00 44 00 00 64 06 90 9f c0 a8 00 77 36 e4 cd 60 04 07 00 50 92 6b 64 51 20 72 8c ec 50 18 01 00 2c 28 00 00 50 4f 53 54 20 2f 75 70 64 61 74 65 20 48 54 54 50 2f 31 2e 30 0d 0a 48 6f 73 74 3a 20 67 61 74 65 77 61 79 2e 77 65 61 74 68 65 72 2e 6f 72 65 67 6f 6e 73 63 69 65 6e 74 69 66 69 63 2e 63 6f 6d 0d 0a 43 6f 6e 74 65 6e 74 2d 4c 65 6e 67 74 68 3a 20 37 35 0d 0a 55 73 65 72 2d 41 67 65 6e 74 3a 20 42 6c 61 63 6b 42 6f 78 2f 31 2e 30 2e 30 38 0d 0a 43 6f 6e 6e 65 63 74 69 6f 6e 3a 20 63 6c 6f 73 65 0d 0a 0d 0a
Feb 17 15:55:20 raspberrypi weewx[809]: interceptor: ServerThread: sniff: timestamp=1487343320.44 pktlen=129 data=00 24 d4 b4 80 5e 00 04 a3 69 03 fe 08 00 45 00 00 73 00 45 00 00 64 06 90 dc c0 a8 00 77 36 e4 cd 60 04 07 00 50 92 6b 64 da 20 72 8c ec 50 18 01 00 67 6e 00 00 6d 61 63 3d 30 30 30 34 61 33 36 39 30 33 66 65 26 69 64 3d 38 34 26 72 69 64 3d 66 33 26 70 77 72 3d 30 26 68 74 72 3d 30 26 63 7a 3d 31 26 6f 68 3d 34 31 26 74 74 72 3d 30 26 6f 74 3d 32 34 2e 35 26 63 68 3d 31 26 70 3d 32
```

```
$Ô´^£iþE±DdÀ¨w6äÍ`PkdQ rìP,(POST /update HTTP/1.0
Host: gateway.weather.oregonscientific.com
Content-Length: 75
User-Agent: BlackBox/1.0.08
Connection: close

$Ô´^£iþEsEdÜÀ¨w6äÍ`PkdÚ rìPgnmac=0004a36903fe&id=84&rid=f3&pwr=0&htr=0&cz=1&oh=41&ttr=0&ot=24.5&ch=1&p=2
```
But the weewx seems still not working. Datas are not refreshed.

I'm getting closer to the goal... any another idea ?

matthewwall, 2017-02-18:
please post more than just two packets.  about 30 or 40 would be nice, but i need at least 5 or 6 to make sure i get the patterns right.

matthewwall, 2017-02-18:
please try 0.28rc1.  the sniffer should now handle POST data.

regis44, 2017-02-18:
It works !! I've just updated with 0.28rc1 and reconfigure the interceptor with the right IP station like that :

```
iface = eth1
pcap_filter = src 192.168.0.119 and dst port 80
```

Now, I will be able to fully enjoy my weather station.

Thank you for your great job.

closed 2017-03-15 by matthewwall