matthewwall/weewx-interceptor #15: Acurite rainfall counter issue

Ketonom, 2017-03-13:
I have an Acurite smartHUB that I'm using with the 01024M package that includes a lightning detector sensor.  While Acurite claims that the smartHUB does not support the lightning detector, it actually appears to be picking it up as just another 5in1 device.  It seems to be reporting the stats to the MyAcurite site as well, but most of the values remain static and even the temperature and humidity are incorrect.

Just the other day, it rained a bit and my actual 5in1 sensor outside registered 0.02 inches of rainfall.  However, weewx kept adding up additional rainfall with each new packet it captured from the 5in1.  After taking a brief look at the code, I believe that this is possibly being caused by the lightning sensor continuously reporting 0.00 inches of rain, which is then resetting the counter variable being used in the driver back to 0.  Thus, the next time the actual 5in1 reports the total rainfall (0.02), weewx sees that as more rainfall and adds it to the total.

Below are some packets I captured by running the driver directly via the command line.  The 00000011 sensor is the real 5in1 and the 00002810 sensor is the lightning detector.  I suspect this could happen even if there were another rain gauge within range of the smartHUB as well.

> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x38', 'sensor_id': '00000011'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x38&sensor=00000011&windspeedmph=5&humidity=93&tempf=39.0&baromin=30.06&battery=normal&rssi=3
> raw packet: {'rssi.00000011.[smarthubID]': 75.0, 'bridge_id.00000011.[smarthubID]': '[smarthubID]', 'sensor_id.00000011.[smarthubID]': '00000011', 'windspeed.00000011.[smarthubID]': 5.0, 'pressure.00000011.[smarthubID]': 30.06, 'dateTime': 1489376978, 'sensor_type.00000011.[smarthubID]': '5N1x38', 'temperature.00000011.[smarthubID]': 39.0, 'dateTime.00000011.[smarthubID]': 1489376978, 'humidity.00000011.[smarthubID]': 93.0, 'usUnits.00000011.[smarthubID]': 1, 'battery.00000011.[smarthubID]': 0, 'usUnits': 1}
> mapped packet: {'txBatteryStatus': 0, 'outTemp': 39.0, 'outHumidity': 93.0, 'dateTime': 1489376978, 'pressure': 30.06, 'windSpeed': 5.0, 'rxCheckPercent': 75.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x38', 'sensor_id': '00002810'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x38&sensor=00002810&windspeedmph=0&humidity=41&tempf=57.7&baromin=30.06&battery=normal&rssi=2
> raw packet: {'dateTime.00002810.[smarthubID]': 1489376986, 'pressure.00002810.[smarthubID]': 30.06, 'usUnits.00002810.[smarthubID]': 1, 'rssi.00002810.[smarthubID]': 50.0, 'humidity.00002810.[smarthubID]': 41.0, 'bridge_id.00002810.[smarthubID]': '[smarthubID]', 'temperature.00002810.[smarthubID]': 57.7, 'windspeed.00002810.[smarthubID]': 0.0, 'sensor_id.00002810.[smarthubID]': '00002810', 'battery.00002810.[smarthubID]': 0, 'sensor_type.00002810.[smarthubID]': '5N1x38', 'dateTime': 1489376986, 'usUnits': 1}
> mapped packet: {'txBatteryStatus': 0, 'outTemp': 57.7, 'outHumidity': 41.0, 'dateTime': 1489376986, 'pressure': 30.06, 'windSpeed': 0.0, 'rxCheckPercent': 50.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x31', 'sensor_id': '00000011'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x31&sensor=00000011&windspeedmph=5&winddir=180&rainin=0.01&dailyrainin=0.01&baromin=30.06&battery=normal&rssi=3
> raw packet: {'rssi.00000011.[smarthubID]': 75.0, 'bridge_id.00000011.[smarthubID]': '[smarthubID]', 'sensor_id.00000011.[smarthubID]': '00000011', 'windspeed.00000011.[smarthubID]': 5.0, 'pressure.00000011.[smarthubID]': 30.06, 'dateTime': 1489376997, 'sensor_type.00000011.[smarthubID]': '5N1x31', 'dateTime.00000011.[smarthubID]': 1489376997, 'usUnits.00000011.[smarthubID]': 1, 'rainfall.00000011.[smarthubID]': None, 'battery.00000011.[smarthubID]': 0, 'usUnits': 1, 'winddir.00000011.[smarthubID]': 180.0}
> mapped packet: {'txBatteryStatus': 0, 'rain': None, 'dateTime': 1489376997, 'windDir': 180.0, 'pressure': 30.06, 'windSpeed': 5.0, 'rxCheckPercent': 75.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x31', 'sensor_id': '00002810'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x31&sensor=00002810&windspeedmph=0&winddir=0&rainin=0.00&dailyrainin=0.00&baromin=30.06&battery=normal&rssi=2
> raw packet: {'pressure.00002810.[smarthubID]': 30.06, 'usUnits.00002810.[smarthubID]': 1, 'dateTime.00002810.[smarthubID]': 1489377004, 'bridge_id.00002810.[smarthubID]': '[smarthubID]', 'rssi.00002810.[smarthubID]': 50.0, 'windspeed.00002810.[smarthubID]': 0.0, 'sensor_id.00002810.[smarthubID]': '00002810', 'battery.00002810.[smarthubID]': 0, 'rainfall.00002810.[smarthubID]': 0.0, 'winddir.00002810.[smarthubID]': 0.0, 'sensor_type.00002810.[smarthubID]': '5N1x31', 'dateTime': 1489377004, 'usUnits': 1}
> mapped packet: {'txBatteryStatus': 0, 'rain': 0.0, 'dateTime': 1489377004, 'windDir': 0.0, 'pressure': 30.06, 'windSpeed': 0.0, 'rxCheckPercent': 50.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x38', 'sensor_id': '00000011'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x38&sensor=00000011&windspeedmph=4&humidity=93&tempf=39.0&baromin=30.06&battery=normal&rssi=3
> raw packet: {'rssi.00000011.[smarthubID]': 75.0, 'bridge_id.00000011.[smarthubID]': '[smarthubID]', 'sensor_id.00000011.[smarthubID]': '00000011', 'windspeed.00000011.[smarthubID]': 4.0, 'pressure.00000011.[smarthubID]': 30.06, 'dateTime': 1489377016, 'sensor_type.00000011.[smarthubID]': '5N1x38', 'temperature.00000011.[smarthubID]': 39.0, 'dateTime.00000011.[smarthubID]': 1489377016, 'humidity.00000011.[smarthubID]': 93.0, 'usUnits.00000011.[smarthubID]': 1, 'battery.00000011.[smarthubID]': 0, 'usUnits': 1}
> mapped packet: {'txBatteryStatus': 0, 'outTemp': 39.0, 'outHumidity': 93.0, 'dateTime': 1489377016, 'pressure': 30.06, 'windSpeed': 4.0, 'rxCheckPercent': 75.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x38', 'sensor_id': '00002810'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x38&sensor=00002810&windspeedmph=0&humidity=41&tempf=57.7&baromin=30.06&battery=normal&rssi=2
> raw packet: {'dateTime.00002810.[smarthubID]': 1489377022, 'pressure.00002810.[smarthubID]': 30.06, 'usUnits.00002810.[smarthubID]': 1, 'rssi.00002810.[smarthubID]': 50.0, 'humidity.00002810.[smarthubID]': 41.0, 'bridge_id.00002810.[smarthubID]': '[smarthubID]', 'temperature.00002810.[smarthubID]': 57.7, 'windspeed.00002810.[smarthubID]': 0.0, 'sensor_id.00002810.[smarthubID]': '00002810', 'battery.00002810.[smarthubID]': 0, 'sensor_type.00002810.[smarthubID]': '5N1x38', 'dateTime': 1489377022, 'usUnits': 1}
> mapped packet: {'txBatteryStatus': 0, 'outTemp': 57.7, 'outHumidity': 41.0, 'dateTime': 1489377022, 'pressure': 30.06, 'windSpeed': 0.0, 'rxCheckPercent': 50.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x31', 'sensor_id': '00000011'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x31&sensor=00000011&windspeedmph=4&winddir=158&rainin=0.01&dailyrainin=0.01&baromin=30.06&battery=normal&rssi=3
> raw packet: {'rssi.00000011.[smarthubID]': 75.0, 'bridge_id.00000011.[smarthubID]': '[smarthubID]', 'sensor_id.00000011.[smarthubID]': '00000011', 'windspeed.00000011.[smarthubID]': 4.0, 'pressure.00000011.[smarthubID]': 30.06, 'dateTime': 1489377035, 'sensor_type.00000011.[smarthubID]': '5N1x31', 'dateTime.00000011.[smarthubID]': 1489377035, 'usUnits.00000011.[smarthubID]': 1, 'rainfall.00000011.[smarthubID]': 0.01, 'battery.00000011.[smarthubID]': 0, 'usUnits': 1, 'winddir.00000011.[smarthubID]': 158.0}
> mapped packet: {'txBatteryStatus': 0, 'rain': 0.01, 'dateTime': 1489377035, 'windDir': 158.0, 'pressure': 30.06, 'windSpeed': 4.0, 'rxCheckPercent': 75.0, 'usUnits': 1}
> 
> identifiers: {'bridge_id': '[smarthubID]', 'sensor_type': '5N1x31', 'sensor_id': '00002810'}
> raw data: dateutc=now&action=updateraw&realtime=1&id=[smarthubID]&mt=5N1x31&sensor=00002810&windspeedmph=0&winddir=0&rainin=0.00&dailyrainin=0.00&baromin=30.06&battery=normal&rssi=2
> raw packet: {'pressure.00002810.[smarthubID]': 30.06, 'usUnits.00002810.[smarthubID]': 1, 'dateTime.00002810.[smarthubID]': 1489377040, 'bridge_id.00002810.[smarthubID]': '[smarthubID]', 'rssi.00002810.[smarthubID]': 50.0, 'windspeed.00002810.[smarthubID]': 0.0, 'sensor_id.00002810.[smarthubID]': '00002810', 'battery.00002810.[smarthubID]': 0, 'rainfall.00002810.[smarthubID]': 0.0, 'winddir.00002810.[smarthubID]': 0.0, 'sensor_type.00002810.[smarthubID]': '5N1x31', 'dateTime': 1489377040, 'usUnits': 1}
> mapped packet: {'txBatteryStatus': 0, 'rain': 0.0, 'dateTime': 1489377040, 'windDir': 0.0, 'pressure': 30.06, 'windSpeed': 0.0, 'rxCheckPercent': 50.0, 'usUnits': 1}
>

matthewwall, 2017-03-14:
the default configuration is to process packets from every sensor.

use the sensor map to restrict your inputs to a single sensor cluster like this:

[Interceptor]
    ...
    [[sensor_map]]
        inTemp = temperature_in.*.*'
        inHumidity = humidity_in.*.*
        outTemp = temperature.00000011.*
        outHumidity = humidity.00000011.*
        windSpeed = windspeed.00000011.*
        windDir = winddir.00000011.*
        rain = rainfall.00000011.*
        txBatteryStatus = battery.00000011.*
        rxCheckPercent = rssi.00000011.*
        extraTemp1 = temperature.00002810.*
        extraHumid1 = humidity.00002810.*

this is assuming that the 5n1 is id 00000011 and the lightning sensor is 00002810

you

Ketonom, 2017-03-14:
Ah, I actually have been doing that, as I have a couple of temp+humidity sensors as well.  For reference, the total rain registered on the display and on myAcurite was 0.02 inches, but weewx had gotten up to 0.14 inches by the time I stopped it within maybe 10-15 minutes, give or take.  Here is the map I have been using, including during the time I recorded those packets:

    [[sensor_map]]
        barometer = pressure.00000011.*
        inTemp = temperature.00010538.*
        outTemp = temperature.00000011.*
        inHumidity = humidity.00010538.*
        outHumidity = humidity.00000011.*
        windSpeed = windspeed.00000011.*
        windDir = winddir.00000011.*
        rain = rainfall.00000011.*
        extraTemp1 = temperature.00013565.*
        extraTemp2 = temperature.00016125.*
        extraHumid1 = humidity.00013565.*
        extraHumid2 = humidity.00016125.*
        rxCheckPercent = rssi.00000011.*
        outTempBatteryStatus = battery.00000011.*
        inTempBatteryStatus = battery.00010538.*

matthewwall, 2017-03-14:
post the weewx log output when debug=1

the output you posted is from running the driver directly, so it only shows the default sensor map, not the sensor map you specified

Ketonom, 2017-03-14:
The attached log starts right before I manually triggered the cup to tip and then shows the rain counter wrap twice due to the lightning sensor, even though it isn't mapped for rain.  I only included the interceptor MainThread output, but if you need to ServerThread output as well, just let me know.

[weewx-debug-excerpt.txt](https://github.com/matthewwall/weewx-interceptor/files/842987/weewx-debug-excerpt.txt)

matthewwall, 2017-03-15:
it is capturing rain only from 00000011, so the problem is not interference from multiple sensors.

the driver should do a delta from the current dailyrainin to the previous dailyrainin, but it looks like that is not happening.

what version of interceptor are you running?

Ketonom, 2017-03-15:
The driver appears to hang if I use the --version parameter, but here is the version from the script file itself:

DRIVER_NAME = 'Interceptor'
DRIVER_VERSION = '0.28rc1'

matthewwall, 2017-03-15:
please update to 0.28.  i added a rain_total to the packets so we should be able to see if there is any difference between delta and total.

matthewwall, 2017-03-15:
please update to 0.29 (commit 46b4ab6)

that should fix your problem.

Ketonom, 2017-03-15:
I stopped weewx, upgraded the interceptor driver, then started it back up.  It now shows both sensors reporting rain and looks like it skipped the initial total value of 0.02 that was already being reported on  on sensor 11 (due to the prior testing) since there was no last rain.  I went ahead and tipped it one more time and got a reading of 0.01 which is now holding steady.  Thanks for your assistance on this!  Great work, as always!

closed 2017-03-15 by Ketonom