matthewwall/weewx-interceptor #17: bug in decode_wu_datetime

noufris, 2017-03-26:
Hello, 
I installed the weewx-interceptor in a debian machine. I try to send this get request: 
curl -s 192.168.1.234:227/weatherstation/updateweatherstation.asp?ID=12345678&PASSWORD=PASSWORD&outtemp=6.3&outhumi=80&dewpoint=3.1&windchill=6.3&winddir=197&windspeed=0.0&windgust=0.0&rainrate=0.0&dailyrain=0.0&weeklyrain=0.0&monthlyrain=0.0&yearlyrain=0.0&light=0.00&UV=1&intemp=19.8&inhumi=46&absbaro=1018.30&relbaro=1018.30&lowbatt=0&dateutc=2016-4-30%2021:5:1&softwaretype=Weather%20logger%20V2.1.9&action=updateraw&realtime=1&rtfreq=5

and nothing happes. The /var/log/syslog is:

engine: Main loop exiting. Shutting engine down.
Mar 26 21:02:44 minoas weewx[6241]: interceptor: MainThread: shutting down server thread
Mar 26 21:02:44 minoas weewx[6241]: interceptor: MainThread: stop tcp server
Mar 26 21:02:45 minoas weewx[6241]: engine: Caught unrecoverable exception in engine:
Mar 26 21:02:45 minoas weewx[6241]:     ****  'int' object has no attribute 'replace'
Mar 26 21:02:45 minoas weewx[6241]:     ****  Traceback (most recent call last):
Mar 26 21:02:45 minoas weewx[6241]:     ****    File "/usr/share/weewx/weewx/engine.py", line 871, in main
Mar 26 21:02:45 minoas weewx[6241]:     ****      engine.run()
Mar 26 21:02:45 minoas weewx[6241]:     ****    File "/usr/share/weewx/weewx/engine.py", line 187, in run
Mar 26 21:02:45 minoas weewx[6241]:     ****      for packet in self.console.genLoopPackets():
Mar 26 21:02:45 minoas weewx[6241]:     ****    File "/usr/share/weewx/user/interceptor.py", line 1937, in genLoopPackets
Mar 26 21:02:45 minoas weewx[6241]:     ****      pkt = self._device.parser.parse(data)
Mar 26 21:02:45 minoas weewx[6241]:     ****    File "/usr/share/weewx/user/interceptor.py", line 988, in parse
Mar 26 21:02:45 minoas weewx[6241]:     ****      data.pop('dateutc', int(time.time() + 0.5)))
Mar 26 21:02:45 minoas weewx[6241]:     ****    File "/usr/share/weewx/user/interceptor.py", line 521, in decode_wu_datetime
Mar 26 21:02:45 minoas weewx[6241]:     ****      s = s.replace("%20", " ")
Mar 26 21:02:45 minoas weewx[6241]:     ****  AttributeError: 'int' object has no attribute 'replace'
Mar 26 21:02:45 minoas weewx[6241]:     ****  Exiting.

Could you please tell me what I do wrong?

matthewwall, 2017-03-26:
please try interceptor v0.30

this will handle integer timestamps.

noufris, 2017-03-26:
Thanks for your response. The problem solved, as it don't throw up any error. But I don't see any change in the webpage. Could you please send me a correct get request to test my system? Maybe I do something wrong. Thanks for your time!

matthewwall, 2017-03-26:
the curl request you made looks ok.  what do you see when you run weewxd directly?  it should spit out whatever you posted using curl.

noufris, 2017-03-26:
I stopped the weewx service and run the weewxd directluy. The anwer is:

LOOP:   2017-03-26 21:57:26 EEST (1490554646) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1490554646, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, rainRate: 0, usUnits: 1, windchill: None
 
I can't see any temp or humidity update.

matthewwall, 2017-03-26:
this is now a matter of your configuration.  please post your configuration and sample loop output at the weewx user forum.

closed 2017-03-26 by matthewwall

noufris, 2017-03-26:
Thanks again for your time!