matthewwall/weewx-interceptor #19: Exception while parsing Rain data from LW301

serjs, 2017-10-04:
Hi!

Trying to use interceptor for working with my LW301, due parsing of Rain sensor I receive exception
```
LOOP:   2017-10-04 18:41:20 Europe (1507142480) altimeter: None, appTemp: None, barometer: None, cloudbase: 1263.27382561, dateTime: 1507142480, dewpoint: 43.5615951673, heatindex: 46.04, humidex: 46.04, inDewpoint: None, maxSolarRad: 575.341723869, outHumidity: 91.0, outTemp: 46.04, pressure: None, rainRate: 0, usUnits: 1, windchill: None
REC:    2017-10-04 18:40:00 Europe (1507142400) altimeter: None, appTemp: 44.5523019125, barometer: None, cloudbase: 1263.27382561, dateTime: 1507142400.0, dewpoint: 43.5615951673, ET: None, heatindex: 46.04, humidex: 46.04, inDewpoint: None, interval: 5, maxSolarRad: 571.847376021, outHumidity: 91.0, outTemp: 46.04, pressure: None, rainRate: 0.0, usUnits: 1, windchill: 46.04, windDir: None, windGust: 0.0, windGustDir: None, windrun: 0.0, windSpeed: 0.0
Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/home/weewx/bin/weewx/engine.py", line 871, in main
    engine.run()
  File "/home/weewx/bin/weewx/engine.py", line 187, in run
    for packet in self.console.genLoopPackets():
  File "/home/weewx/bin/user/interceptor.py", line 1976, in genLoopPackets
    pkt = self._device.parser.parse(data)
  File "/home/weewx/bin/user/interceptor.py", line 1175, in parse
    last_rain = self._last_rain[sensor_id]
KeyError: '1:bc'
```

Looks like bc it is a sensor ID and 1 it is a channel, here example request from tcpflow
```
mac=<erased>&id=82&rid=bc&pwr=0&rro=0&rr=0.00&rfa=61.984&ch=1&p=1GET /weatherstation/WXDailyHistory.asp?ID=ISOVYAKI3&month=10&day=4&year=2017&format=1 HTTP/1.1
Accept-Encoding: identity
Host: www.wunderground.com
Connection: close
User-Agent: Python-urllib/2.7
```

matthewwall, 2017-10-04:
fixed at commit 28d02b6

closed 2017-10-04 by matthewwall

serjs, 2017-10-05:
Thank you for quick fix!

5+ hours - everything is fine