matthewwall/weewx-interceptor #20: Driver and VPS

Eater4, 2017-10-07:
Is it possible for this driver to work on a vps? I have a vps that has weewx running on it and and would like to get the data from my lw301 at home. How would I go about doing this?

matthewwall, 2017-10-07:
perhaps the easiest way to do it is to sniff the traffic from the lw301 on the router at your home, and have the router send that traffic to the weewx instance on your vps.  you can do this with a simple firewall rule on the router, or by running tcpdump on the router.

another way to do it is to hijack dns - configure your router to return the ip address of your vps instead of the ip address of weather underground (or whatever other host the lw301 tries to send to)

yet another way is to use a raspberry pi as a wifi access point, connect the lw301 to the pi's wifi, then run a sniffer on the pi.  basically this has the same options as the router configurations, but instead of doing it on your router, you do it on the pi.

Eater4, 2017-10-08:
How would I go about doing the last one of the raspberry pi only has one
Ethernet port? Get one that has two?

matthewwall, 2017-10-08:
plug the pi ethernet port into your local network.  configure the pi wifi to be an access point.  configure the lw301 to use the wifi ssid that the pi is advertising.

then all you have to do is run a sniffer on the pi, and have the sniffer send to your vps.

of course, you could just as easily run weewx on the pi, and let the interceptor driver do the sniffing.  then rsync the weewx database to the vps, or let weewx push the reports to the vps, or whatever.

generally speaking it is a good idea to have weewx as close to the weather hardware as possible (logically and physically).  that way you minimize points of failure, and you end up with a local copy of the data, even if the rest of the world goes away.

matthewwall, 2017-10-08:
note that this requires a pi3 (they have built-in wifi).

if you use a pi2, then you'll have to get a usb-wifi dongle or some other combination of hardware.

matthewwall, 2017-12-20:
closing this due to no activity

closed 2017-12-20 by matthewwall