matthewwall/weewx-interceptor #21: Rain Rate is wrong lw30x

cryptooth, 2017-10-28:
I'm running latest version of weewx and interceptor. My sensor is lw301.

Real rain rate is 2.1mm/hr on OSAnywhere SW, but weewx is reporting 0.08 mm/hr.

Any advice?

matthewwall, 2017-10-28:
does the value for rainRate bounce around, or is it pretty consistent?

you might want to do this in weewx.conf:

```
[StdWXCalculate]
    ...
    rainRate = software
```

the lw30x reports rain rate, but not in every packet.  so weewx will calculate a rainRate for every packet that does not contain a rainRate.  setting rainRate=software will make weewx calculate the rainRate for every package.

the weewx algorithm for rainRate is probably not the same as the algorithm used by the lw30x, so you probably will never get an exact match.

finally, it is possible that the weewx-interceptor driver is interpreting the lw30x output incorrectly.  the driver interprets 'rr' as the rain rate.  if you could capture some of the raw output from the lw30x we could verify this.  set debug=1 then restart weewx.  then wait for a rain event.

cryptooth, 2017-10-29:
Thank you for quick reply. Istanbul is quite rainy these days. I will try to capture data and send to you.

Belek

matthewwall, 2017-12-20:
i think this has been fixed.  please create a new issue if there is still a problem.

closed 2017-12-20 by matthewwall