matthewwall/weewx-interceptor #22: Sniffing GW1000u not implemented in interceptor.py version 0.38?

ohioflowers, 2017-11-22:
In the current interceptor.py file I note:

#FIXME: implement packet sniffing mode for gw1000u
#FIXME: implement standalone option to detect gw1000u broadcasts and configure

Does this mean I cannot use my gw1000u in sniff mode at present?  Disappointing, as that leaves only listen mode with destination address spoofed by GAS (it works) so no lacrosse alerts possible.

matthewwall, 2017-11-22:
correct.

sniffing should be pretty straightforward to write now that the listen mode is working.  the only tricky part is to ensure that the packets are reassembled properly before decoding.

that will require some testing and iteration

ohioflowers, 2017-11-22:
If desired, I can help with testing.  While I don’t know Python I have over 20 years of coding experience in various languages and I’ve done 3 installs of interceptor on pi 3 and macos trying to figure out why sniffing doesn’t work.  I couldn’t figure out how to spoof the destination address back to the original with dd-wrt so that was my last hope.

matthewwall, 2017-11-22:
if you can get some packet captures of the communications between the gw1000u and lacrosse alerts, i could use those to implement the sniffer.  i probably have a bunch of old captures sitting around that we used to reverse engineer the protocol, but it would be better to work with something more recent.

i am backlogged right now, but i could probably work on this by late december

ohioflowers, 2017-11-22:
I’ll put something together and send it to you.  Assume raw tcpdump files will be ok.

matthewwall, 2017-11-22:
given your experience, you could probably do the coding.  happy to help if you get the itch before i can get to it.

raw tcpdump is fine.  context is helpful, i.e., how tcpdump was invoked, what is the gw1000u ip address/macaddr, serial number, etc.

the interceptor is already set up to do sniffing, but it expects the RESTful-like GET/POST traffic that most weather stations use.  the gw1000u uses PUT and a very specific client-server interaction.  when sniffing, you won't care about the interaction, but parsing the PUT will be different from the parsing/packetization of the GET/POST traffic.

once the PUTs are properly captured and assembled, it should just be a matter of looking for the pkt_type identifiers then feeding to the GW1000U.Parser.

the driver has some testing capability built-in.  when you run the driver directly you can ask it to parse test packets and other operations.  that way you can validate the driver without the rest of the weewx plumbing.  it makes for much faster development and debugging.
```
cd /home/weewx
PYTHONPATH=bin python bin/user/interceptor.py --help
```

ohioflowers, 2017-11-22:
Well no itch but I do like problem solving, particularly with some minimal help so maybe I’ll get to looking at it.  Right now I’m running it with the acurite-bridge device and getting all kinds of log entries instead of empty queue messages once I commented out my gw1000u serial number.

At least it looks like progress.

Thanks for the comments.  I tried direct but sniff mode always reverted to listen.

ohioflowers, 2017-11-22:
Well just to pique your interest, the following log entry resulted from using the accurate-bridge device:

Nov 22 16:32:41 raspberrypi weewx[12551]: interceptor: ServerThread: sniff: timestamp=1511386361.5 pktlen=239 data=68 7f 74 a3 95 eb 00 1d 8c 09 8a 5c 08 00 45 00 00 e1 00 67 00 00 ff 06 cd b0 c0 a8 01 48 34 04 f7 0a 06 47 00 50 00 00 4a 01 92 36 7d c4 50 18 16 d0 c3 b6 00 00 50 55 54 20 2f 72 65 71 75 65 73 74 2e 62 72 65 71 20 48 54 54 50 2f 31 2e 31 0d 0a 48 6f 73 74 3a 20 62 6f 78 2e 77 65 61 74 68 65 72 64 69 72 65 63 74 2e 63 6f 6d 0d 0a 43 6f 6e 6e 65 63 74 69 6f 6e 3a 20 63 6c 6f 73 65 0d 0a 48 54 54 50 5f 49 44 45 4e 54 49 46 59 3a 20 38 30 30 39 38 41 35 43 3a 30 30 3a 34 30 32 38 43 36 41 33 30 38 43 32 34 32 38 43 3a 37 30 0d 0a 43 6f 6e 74 65 6e 74 2d 54 79 70 65 3a 20 61 70 70 6c 69 63 61 74 69 6f 6e 2f 6f 63 74 65 74 2d 73 74 72 65 61 6d 0d 0a 43 6f 6e 74 65 6e 74 2d 4c 65 6e 67 74 68 3a 20 30 0d 0a 0d 0a

I’m too rusty to read the packets other than the ip addresses without a reference but it looks pretty good to me and all the debug=1 sniff references seem to be in the log, as well.  src host 192.168.1.72 dst host 52.4.247.10 dst port 80 next hop 192.169.1.20.

I don’t think underground will know what to do with it but I might give it a try anyway after I figure out what weewx is doing with it.

matthewwall, 2017-11-22:
you might have read it already, but see section 'How to run the driver directly' in the interceptor readme.

you will want something like this:

```
PYTHONPATH=bin python bin/user/interceptor.py --device=gw1000u --mode=sniff --iface=eth0 --filter="src 192.168.0.4 and dst port 80" --debug
```

that should give you more to work with, and without the weewx overhead.  that is how i test the driver.  once you get good data from that, you'll know you're getting good data in weewx.

wu won't know what to make of the gw1000u conversations