matthewwall/weewx-interceptor #23: Published record delay

kan6526, 2017-12-17:
Hi
My wifi weather station is foshk [wh2950](http://www.foshk.com/Wifi_Weather_Station/WH2950.html)
I installed weewx(3.8.0a2) on the pc and used the interceptor(0.39) to recover the data
I found that pushing the data will be delayed for a long time.

Thanks for the help!!
-----------------------weewx.conf---------------
[Simulator]

    loop_interval = 2.5
     mode = simulator
    driver = weewx.drivers.simulator

[StdArchive]

    archive_interval = 60
    record_generation = software
    loop_hilo = True
    data_binding = wx_binding

---------------------syslog---------------------

Dec 17 12:36:04  weewx[13162]: engine: Finished loading service user.windguru.WindGuru
Dec 17 13:04:04  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:37:00 CST (1513485420)
Dec 17 13:09:31  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:38:00 CST (1513485480)
Dec 17 13:14:34  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:39:00 CST (1513485540)
Dec 17 13:18:34  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:40:00 CST (1513485600)
Dec 17 13:22:35  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:41:00 CST (1513485660)
Dec 17 13:28:21  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:42:00 CST (1513485720)
Dec 17 13:32:32  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:43:00 CST (1513485780)
Dec 17 13:36:52  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:44:00 CST (1513485840)
Dec 17 13:40:52  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:45:00 CST (1513485900)
Dec 17 13:46:28  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:46:00 CST (1513485960)
Dec 17 13:49:41  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:47:00 CST (1513486020)
Dec 17 13:53:31  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:48:00 CST (1513486080)
Dec 17 13:57:08  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:49:00 CST (1513486140)
Dec 17 14:01:41  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:50:00 CST (1513486200)
Dec 17 14:05:56  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:51:00 CST (1513486260)
Dec 17 14:10:35  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:52:00 CST (1513486320)
Dec 17 14:14:44  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:53:00 CST (1513486380)
Dec 17 14:19:01  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:54:00 CST (1513486440)
Dec 17 14:22:44  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:55:00 CST (1513486500)
Dec 17 14:27:48  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:56:00 CST (1513486560)
Dec 17 14:33:08  weewx[13162]: restx: WindGuru: Published record 2017-12-17 12:57:00 CST (1513486620)

matthewwall, 2017-12-20:
is that raw log output?  it looks like it has been modified.

please set debug=1 in weewx.conf

then please post the log from the time that weewx start up til after 10 minutes or so.

kan6526, 2017-12-21:
Hi 
Published  is 5 minutes, how to change into 1 minute?

my syslog.log
https://drive.google.com/file/d/15JTWY3FPA2yfvUydtry4TLhCTwERKDpK/view

matthewwall, 2018-05-24:
the archive interval is specified in the weewx configuration file as 'archive_interval'

closed 2018-05-24 by matthewwall