matthewwall/weewx-interceptor #24: Import of driver failed???

couchfisher, 2018-01-01:
Hopefully someone can help. I have the Lacrosse Weather Station and when I initially set it up, everything worked. I could see the log data and information in file:///var/www/html/weewx/index.html. 
After I modified the weewx.conf for weather underground it no longer finds the driver. I cannot seem to get it to work. If I uninstall everything and re-install it works again, only when I first start weewx. ??

After initial setup:

pi@raspberrypi:~ $ sudo /etc/init.d/weewx start
[ ok ] Starting weewx (via systemctl): weewx.service.
pi@raspberrypi:~ $ sudo tail -f /var/log/syslog
Jan  1 17:52:54 raspberrypi dbus-daemon[563]: Successfully activated service 'ca.desrt.dconf'
Jan  1 17:52:54 raspberrypi dbus[348]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan  1 17:52:54 raspberrypi systemd[1]: Started Hostname Service.
Jan  1 17:53:02 raspberrypi gvfsd[619]: dbus_mount_reply: Error from org.gtk.vfs.Mountable.mount(): Failed to retrieve share list from server: Connection refused
Jan  1 17:53:02 raspberrypi gvfsd-network[2546]: Couldn't create directory monitor on smb://x-gnome-default-workgroup/. Error: The specified location is not mounted
Jan  1 17:55:16 raspberrypi weewx[2498]: manager: Added record 2018-01-01 17:55:00 UTC (1514829300) to database 'weewx.sdb'
Jan  1 17:55:16 raspberrypi weewx[2498]: manager: Added record 2018-01-01 17:55:00 UTC (1514829300) to daily summary in 'weewx.sdb'
Jan  1 17:55:21 raspberrypi weewx[2498]: cheetahgenerator: Generated 14 files for report StandardReport in 4.20 seconds
Jan  1 17:55:26 raspberrypi weewx[2498]: imagegenerator: Generated 49 images for StandardReport in 5.03 seconds
Jan  1 17:55:26 raspberrypi weewx[2498]: copygenerator: copied 9 files to /var/www/html/weewx
Jan  1 18:00:16 raspberrypi weewx[2498]: manager: Added record 2018-01-01 18:00:00 UTC (1514829600) to database 'weewx.sdb'
Jan  1 18:00:16 raspberrypi weewx[2498]: manager: Added record 2018-01-01 18:00:00 UTC (1514829600) to daily summary in 'weewx.sdb'
Jan  1 18:00:18 raspberrypi weewx[2498]: cheetahgenerator: Generated 14 files for report StandardReport in 1.29 seconds
Jan  1 18:00:21 raspberrypi weewx[2498]: imagegenerator: Generated 36 images for StandardReport in 3.48 seconds
Jan  1 18:00:21 raspberrypi weewx[2498]: copygenerator: copied 0 files to /var/www/html/weewx
^C
pi@raspberrypi:~ $ sudo /etc/init.d/weewx stop

After adding Weather Underground informaiton:

pi@raspberrypi:~ $ sudo /etc/init.d/weewx start
[ ok ] Starting weewx (via systemctl): weewx.service.
pi@raspberrypi:~ $ sudo tail -f /var/log/syslog
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: sensor map: None
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: using station serial 0000000000000000
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: using ping interval 240s
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: using sensor interval 1m
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: using history interval 1 (5m)
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: mode is listen
Jan  1 18:06:34 raspberrypi weewx[2935]: interceptor: MainThread: listen on 192.168.2.22:80
Jan  1 18:06:34 raspberrypi weewx[2935]: import of driver failed: [Errno 99] Cannot assign requested address (<class 'socket.error'>)
Jan  1 18:06:34 raspberrypi weewx[2935]: engine: Unable to load driver: [Errno 99] Cannot assign requested address
Jan  1 18:06:34 raspberrypi weewx[2935]:     ****  Exiting...
^C

matthewwall, 2018-01-01:
is there something else listening on port 80?

are you certain that weewx was fully shut down after you did '/etc/init.d/weewx stop'?  do 'ps ax | grep wee' to see if weewx is still running.  do 'netstat -a | grep http' to see if port 80 is in use.

matthewwall, 2018-05-24:
closing this because no activity.  if it is still a problem, please post to the weewx-user group.  use this issue list once we isolate to a specific bug in the weewx-interceptor code.

closed 2018-05-24 by matthewwall

RightRozz, 2019-03-02:
Same issue here...  Nothing on netstat command.

Mar  1 18:33:40 raspberrypi weewx[13509]: engine: Using configuration file /etc/weewx/weewx.conf
Mar  1 18:33:40 raspberrypi weewx[13509]: engine: Debug is 1
Mar  1 18:33:40 raspberrypi weewx[13509]: engine: Initializing engine
Mar  1 18:33:40 raspberrypi weewx[13509]: engine: Loading station type Interceptor (user.interceptor)
Mar  1 18:33:41 raspberrypi weewx[13509]: interceptor: MainThread: driver version is 0.45
Mar  1 18:33:41 raspberrypi weewx[13509]: interceptor: MainThread: device type: acurite-bridge
Mar  1 18:33:41 raspberrypi weewx[13509]: interceptor: MainThread: sensor map: None
Mar  1 18:33:41 raspberrypi weewx[13509]: interceptor: MainThread: mode is listen
Mar  1 18:33:41 raspberrypi weewx[13509]: interceptor: MainThread: listen on 192.168.0.102:80
Mar  1 18:33:41 raspberrypi weewx[13509]: import of driver failed: [Errno 99] Cannot assign requested address (<class 'socket.error'>)
Mar  1 18:33:41 raspberrypi weewx[13509]: engine: Unable to load driver: [Errno 99] Cannot assign requested address
Mar  1 18:33:41 raspberrypi weewx[13509]:     ****  Exiting...

matthewwall, 2019-03-02:
are you running weewx as root?

Danie10, 2019-03-02:
Pity no-one has responded with whether they came right or noty. I'm getting the same errors but am running weewx as root and port 80 seems to have LISTEN active. If I navigate to that bride IP address and port, the bridge's web page comes up. Here are my outputs:

pi@raspberrypi:~ $ sudo tail -f /var/log/syslog
Mar  2 10:10:52 raspberrypi weewx[1211]: engine: Using configuration file /etc/weewx/weewx.conf
Mar  2 10:10:52 raspberrypi weewx[1211]: engine: Loading station type Interceptor (user.interceptor)
Mar  2 10:10:52 raspberrypi weewx[1211]: interceptor: MainThread: driver version is 0.45
Mar  2 10:10:52 raspberrypi weewx[1211]: interceptor: MainThread: device type: acurite-bridge
Mar  2 10:10:52 raspberrypi weewx[1211]: interceptor: MainThread: sensor map: None
Mar  2 10:10:52 raspberrypi weewx[1211]: interceptor: MainThread: mode is listen
Mar  2 10:10:52 raspberrypi weewx[1211]: interceptor: MainThread: listen on 192.168.1.223:80
Mar  2 10:10:52 raspberrypi weewx[1211]: import of driver failed: [Errno 99] Cannot assign requested address (<class 'socket.error'>)
Mar  2 10:10:52 raspberrypi weewx[1211]: engine: Unable to load driver: [Errno 99] Cannot assign requested address
Mar  2 10:10:52 raspberrypi weewx[1211]:     ****  Exiting...
^C
pi@raspberrypi:~ $ netstat -a | grep http
tcp6       0      0 [::]:http               [::]:*                  LISTEN

RightRozz, 2019-03-02:
Did you not see my edit Danie?  The Listen IP in the config file should be the ip of the weewx computer, NOT the Acurite Bridge's ip.  After I did this I had to configure my router to point traffic going to acurite servers to my weewx computer.  It was a simple DNS configuration.

Danie10, 2019-03-02:
Thanks I did miss your edit. I'll see if I can set up that DNS config change.

Danie10, 2019-03-02:
@RightRozz after a bit of a struggle I think I got the DNS hijacking to work. If I now ping hubapi.myacurite.com it returns the IP address of my weemx computer (192.168.1.85). But I'm still seeing the error below.

Mar  3 00:05:44 raspberrypi weewx[9178]: engine: Using configuration file /etc/weewx/weewx.conf
Mar  3 00:05:44 raspberrypi weewx[9178]: engine: Loading station type Interceptor (user.interceptor)
Mar  3 00:05:44 raspberrypi weewx[9178]: interceptor: MainThread: driver version is 0.45
Mar  3 00:05:44 raspberrypi weewx[9178]: interceptor: MainThread: device type: acurite-bridge
Mar  3 00:05:44 raspberrypi weewx[9178]: interceptor: MainThread: sensor map: None
Mar  3 00:05:44 raspberrypi weewx[9178]: interceptor: MainThread: mode is listen
Mar  3 00:05:44 raspberrypi weewx[9178]: interceptor: MainThread: listen on 192.168.1.85:80
Mar  3 00:05:44 raspberrypi weewx[9178]: import of driver failed: [Errno 98] Address already in use (<class 'socket.error'>)
Mar  3 00:05:44 raspberrypi weewx[9178]: engine: Unable to load driver: [Errno 98] Address already in use
Mar  3 00:05:44 raspberrypi weewx[9178]:     ****  Exiting...

matthewwall, 2019-03-02:
some other process is already listening on port 80.  kill that other process before you start weewx.

Danie10, 2019-03-03:
Thanks @matthewwall that makes sense yes. I did find ip6 listening on that port but after disabling it and rebooting I still get the error so something is still there. It seems to be the essence of what this error is about. I ran the command again to check what is listening but it looks like ip4 TCP? Not sure if it is my Apache server that is serving the webpage? Maybe I should try changing the Apache server to serving on port 8080?

netstat -tulpn
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5900            0.0.0.0:*               LISTEN      -                   
udp        0      0 0.0.0.0:68              0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           -                   
udp        0      0 0.0.0.0:33028           0.0.0.0:*

Danie10, 2019-03-03:
Changed the Apache server listening port to 8080 and it is now all working fine, no errors, and data pulling through fine:

Edit your port by running this command
sudo nano /etc/apache2/ports.conf

Change Listen 80 to 8000
To exit nano editor, press ctrl + x then y to save

Then restart apache
sudo service apache2 restart

larduino, 2019-05-30:
Thanks @Danie10   I made those changes to apache and it got rid of the problem