matthewwall/weewx-interceptor #28: baromin should be mapped to pressure

mattti, 2018-05-27:
Hi!

I have the WH2600 station and use it with the interceptor plugin 0.39.
The problem is, that the "absolute pressure" value of the station gets mapped to "barometer" in weewx.

It is reported as "baromin", so I think, "baromin" should be mapped to "pressure" instead of "barometer".

`GET /weatherstation/updateweatherstation.php?ID=bla&PASSWORD=bla&tempf=64.0&humidity=79&dewptf=57.4&windchillf=64.0&winddir=248&windspeedmph=0.00&windgustmph=0.00&rainin=0.00&dailyrainin=0.00&weeklyrainin=0.00&monthlyrainin=0.00&yearlyrainin=0.00&solarradiation=102.63&UV=1&indoortempf=72.1&indoorhumidity=66&baromin=28.91&lowbatt=0&dateutc=2018-5-27%205:46:19&softwaretype=WH2600GEN_V2.2.5&action=updateraw&realtime=1&rtfreq=5 HTTP/1.0
`

matthewwall, 2018-05-28:
this has gone back and forth - apparently the firmware on some stations sends station pressure for 'baromin', but other firmware/stations send barometer.

have you verified that the wh2600 sends station pressure and not pressure adjusted to sea level?

if that is the case, then we will have to make a special case for that station and firmware version.

mattti, 2018-05-28:
Yes, I have verified that the station posts station pressure as baromin.
For now, I have patched my interceptor.py to reflect this and everything works now as expected.

line 955:
`            'baromin': 'pressure',`

So, no pressure  ;-) (pun not intended)

BTW, thanks for the software :-)

closed 2018-08-31 by matthewwall