matthewwall/weewx-interceptor #31: LW301 reporting rain in inches rather than mm

elandant, 2018-06-17:
I have set up an Oregon LW301 attached to temp/humidity/barometer, wind and rain sensors and run weewx interceptor in listen mode. I am successfully reporting data to Wunderground but after rain yesterday and overnight, the figures seem drastically under-reported - 0.79mm for today when we've actually had just under 20mm since midnight according to the Oregon base station.

Looking at the logs in debug mode it looks like the rain rate being captured from the Oregon would correlate with inches rather than mm. A conversion of the 0.79 accumulated that has been reported to Wunderground from inches to mm is quite accurate.

Is the interceptor doing any conversion or is the problem likely to lie in weewx / my settings? Weewx is configured as metric from setup, but [StdConvert] target_unit = US. All other reported values to Wunderground outside of rainfall/rain rate are accurate.

[image]

matthewwall, 2018-08-31:
do your wind values look ok?  you'll have to look at wind speeds that are greater than 10 miles per hour to be able to tell.  i would not be surprised if the sensors report values in different unit systems.

as of weewx-interceptor v0.41, the packets are treated as METRICWX, which means temperatures in C, wind speeds in m/s, and rainfall in mm.

run weewx directly (e.g., `weewxd /etc/weewx/weewx.conf`) so that you can see the LOOP data directly.  then tip the bucket a single time.  see what is reported in a single LOOP packet.

matthewwall, 2018-12-30:
fixed at commit 8b4f3a9

apparently OS uses inches for rain, m/s for wind speeds, and C for temperatures.  this seems to match the units reported by wmr300, wmr100, and wmr200 hardware.

closed 2018-12-30 by matthewwall