matthewwall/weewx-interceptor #32: interceptor: MainThread: shutting down server thread when adding multiple sensors

richpuch, 2018-06-19:
Interceptor driver is working fine with my 5-1 Acurite Internet Bridge, I have tried to add a tower sensor and editted the following the config settings in weewx.conf so it looked like the following:

[Interceptor]
    # This section is for the network traffic interceptor driver.

    # The driver to use:
    driver = user.interceptor

    # Specify the hardware device to capture.  Options include:
    #   acurite-bridge - acurite internet bridge
    #   observer - fine offset WH2600/HP1000/HP1003, aka 'observer'
    #   lw30x - oregon scientific LW301/LW302
    #   lacrosse-bridge - lacrosse GW1000U/C84612 internet bridge
    device_type = acurite-bridge
    port = 9999

    [[sensor_map]]
        # 5n1
        pressure = pressure.00004032.*
        outTemp = temperature.00004032.*
        outHumidity = humidity.00004032.*
        windSpeed = windspeed.00004032.*
        windDir = winddir.00004032.*
        rain = rainfall.00004032.*
        # outside tower
    #    inTemp = temperature.00014849.*
    #    inHumidity = humidity.00014849.*

When I uncomment inTemp and inHumidity I get intermittent failures and loss of data collection.  Once I remove the extra sensor from the config, weewx runs with no issues.  Running in Debgu mode, I see no real errors, all I catch is:

interceptor: MainThread: shutting down server thread

Which at the point when this error occurs weewx stops for a few minutes and then starts up again.

Running the following versions:
interceptor: MainThread: driver version is 0.39
engine: Initializing weewx version 3.8.0

Any help would be apreciated...my temp sensor in the 5-1 is getting hot and want to add another sensor which is inside a heat shield.

Rich

[debug_RP.txt](https://github.com/matthewwall/weewx-interceptor/files/2115176/debug_RP.txt)

matthewwall, 2018-08-31:
please try the following:

1) shut down weeewx.  make sure that no weewx processes are running.

2) set debug=1 in your weewx.conf file

3) enable the inTemp and inHumidity mappings in your weewx.conf file

4) start weewx

5) let weewx run for at least 3 archive intervals

6) post the log file from just before you started weewx until after the 3rd archive interval

richpuch, 2018-09-07:
Matthew,

Sorry for late response, your notification got lost in spam, thanks for looking into this.  The weird thing is it works fine for a bit and then start throwing out errors.  I'm attaching a jpeg which shows when I lost reporting from weewx and then my /var/log/messages file along with the /var/log/debug in that time frame.

[log-messages.txt](https://github.com/matthewwall/weewx-interceptor/files/2362089/log-messages.txt)
[log-debug.txt](https://github.com/matthewwall/weewx-interceptor/files/2362090/log-debug.txt)
[image]
With the attached logs, it looks like problem started at 13:48 and ended by 13:55.

If I run from just my 5N1 I don't get any errors, one I add the secondary sensor is when I get the drops.

Thanks for your time.
Rich

richpuch, 2018-09-17:
I must have mistakenly closed this issue.

matthewwall, 2018-09-17:
why two different log files?  splitting your logs by priority is really confusing and makes debugging much more difficult.  if you must split logs, then put all the weewx stuff (debug and non-debug) into a single file.  but for diagnosing problems it is far easier to just dump everything into /var/log/messages (that way you can see if there are patterns in weewx that correspond to system events).

one of your skins has problems.  the file wdSearchX3.py has problems at the code 'trend_60_hpa = curr_baro_hpa - baro_60_hpa'.  i think gary fixed that some time ago.  anyway, it is not causing any disruption to the data collection, it just means that the report fails.

you said that simply adding mappings for two observations causes the interceptor to stop collecting data.  but i do not see that in the logs.

please disable all of the wd stuff then run the test i described above.

richpuch, 2018-10-17:
Matthew,

Just ran weewx for a bit and had a few hicups....you can see in the attached log file (Oct 17 14:07).  At this point, I don't log any info to my DB from my 5N1 station, but the Tower sensor keeps logging fine.  

This only happens once I try to log from the Tower too...if I'm just on the 5N1 then no problem.  

File "/home/weewx/bin/user/wdSearchX3.py", line 1554, in get_extension_list
Oct 17 14:07:28 weatherPi weewx[20160]:         ****      trend_60_hpa = curr_baro_hpa - baro_60_hpa
Oct 17 14:07:28 weatherPi weewx[20160]:         ****  TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'

Is it because I have the two barometers from the 5N1 and Tower...I only define one in weewx.conf.

Your interceptor program seems to keep collecting data....I'm failing on the  wdSearchX3.py...this error only occurs when I try to include the tower in my weewx.conf.  If I pull the tower out of weewx.conf...it all is fine.  Any ideas?

[error.txt](https://github.com/matthewwall/weewx-interceptor/files/2488917/error.txt)

richpuch, 2018-10-25:
Matthew,

Thanks for looking into this...I [posted](https://groups.google.com/forum/#!topic/weewx-user/wO9O67SLfgQ) on the weewx forum and Gary fixed the code, you were right it was the wdSearchX3.py file. 

https://groups.google.com/forum/#!topic/weewx-user/wO9O67SLfgQ

Thanks again
Rich

closed 2018-10-25 by richpuch