matthewwall/weewx-interceptor #33: Sniff mode with bridge connected into rPi

holyspidoo, 2018-10-16:
I'm trying to get the smarthub working by connecting it directly into the rPi ethernet. The rPi is connected to rest of LAN via wifi. Looking at the readme, it says:

```
Example: Acurite Bridge in sniff mode
1) The interceptor must be running on a computer that can see traffic from the
   acurite bridge.  For example, run weewx on a raspberry pi that has both a
   wired network interface and a wifi network interface.  Connect the acurite
   bridge to the wired interface, and connect the rpi wireless to the local
   wireless network.

[Interceptor]
    driver = user.interceptor
    device_type = acurite-bridge
    mode = sniff
    iface = eth0
    pcap_filter = src X.X.X.X and dst port 80
```

Isn't there some additional setup needed in this scenario? Like DHCP on the rPi in order to know what X.X.X.X will be? I feel like I'm missing a step.

holyspidoo, 2018-10-28:
This is what was needed to get this to work:
https://github.com/arpitjindal97/raspbian-recipes/blob/master/wifi-to-eth-route.sh

That makes it possible to use sniff mode on the Raspberry Pi

matthewwall, 2018-12-30:
added notes to readme about routing between rpi interfaces at commit 456d2d4

closed 2018-12-30 by matthewwall