matthewwall/weewx-interceptor #34: AMBWeather Invalid Literal

gary-hammer, 2018-10-19:
I'm running an Ambient Weather 2902A and the packets sent are not correctly intercepted using 0.40, or 0.41.

The log shows this:
`Oct 19 16:25:57 ftp weewx[306]: interceptor: MainThread: parse failed for stationtype=AMBWeatherV4.0.2&PASSKEY=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX&dateutc=2018-10-19+20:25:24&winddir=108&windspeedmph=0.9&windgustmph=1.1&maxdailygust=10.3&tempf=58.3&hourlyrainin=0.00&eventrainin=0.00&dailyrainin=0.00&weeklyrainin=0.21&monthlyrainin=4.18&totalrainin=31.25&baromrelin=30.08&baromabsin=29.54&humidity=50&tempinf=71.6&humidityin=38&uv=0&solarradiation=48.93 HTTP/1.1#015#012Host: api.ambientweather.net#015#012Connection: close#015#012User-Agent: ESP8266#015#012#015#012: invalid literal for float(): 48.93 HTTP/1.1#015#012Host: api.ambientweather.net#015#012Connection: close#015#012User-Agent: ESP8266`

`Oct 19 16:26:29 ftp weewx[306]: interceptor: MainThread: parse failed for stationtype=AMBWeatherV4.0.2&PASSKEY=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX&dateutc=2018-10-19+20:26:26&winddir=81&windspeedmph=0.7&windgustmph=1.1&maxdailygust=10.3&tempf=58.1&hourlyrainin=0.00&eventrainin=0.00&dailyrainin=0.00&weeklyrainin=0.21&monthlyrainin=4.18&totalrainin=31.25&baromrelin=30.09&baromabsin=29.55&humidity=50&tempinf=71.6&humidityin=39&uv=0&solarradiation=48.07 HTTP/1.1#015#012Host: api.ambientweather.net#015#012Connection: close#015#012User-Agent: ESP8266#015#012#015#012: invalid literal for float(): 48.07 HTTP/1.1#015#012Host: api.ambientweather.net#015#012Connection: close#015#012User-Agent: ESP8266`

I have replaced the passkey with an equivalent length string. 

If there is any other information you want, just let me know.

matthewwall, 2018-12-30:
did you get this to work?  if not, could you try with the latest interceptor (currently 0.43), and run it with debug enabled.  also, i need to know whether you were running in sniff mode or listen mode, so please post the [Interceptor] section of your weewx configuration.

it looks like you are getting multiple packets in a single connection string.

gary-hammer, 2018-12-30:
I was on 0.41.
When I installed 0.43 I am unable to start weewx.
`Dec 30 16:30:18 ftp systemd[1]: Started weewx weather system.
Dec 30 16:30:18 ftp weewx[24257]: engine: Initializing weewx version 3.8.2
Dec 30 16:30:18 ftp weewx[24257]: engine: Using Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15) #012[GCC 7.3.0]
Dec 30 16:30:18 ftp weewx[24257]: engine: Platform Linux-4.15.0-43-generic-x86_64-with-LinuxMint-19.1-tessa
Dec 30 16:30:18 ftp weewx[24257]: engine: Locale is 'en_US.UTF-8'
Dec 30 16:30:18 ftp weewx[24257]: engine: pid file is /var/run/weewx.pid
Dec 30 16:30:18 ftp weewx[24261]: engine: Using configuration file /home/weewx/weewx.conf
Dec 30 16:30:18 ftp weewx[24261]: engine: Loading station type Interceptor (user.interceptor)
Dec 30 16:30:18 ftp weewx[24261]: interceptor: MainThread: driver version is 0.43
Dec 30 16:30:18 ftp weewx[24261]: interceptor: MainThread: device type: observer
Dec 30 16:30:18 ftp weewx[24261]: interceptor: MainThread: sensor map: None
Dec 30 16:30:18 ftp weewx[24261]: interceptor: MainThread: mode is sniff
Dec 30 16:30:18 ftp weewx[24261]: interceptor: MainThread: sniff iface=eth0 promiscuous=0
Dec 30 16:30:18 ftp weewx[24261]: interceptor: MainThread: sniff filter 'src 10.10.100.121 and dst port 80'
Dec 30 16:30:18 ftp weewx[24261]: import of driver failed: pcapObject must be initialized via open_live(), open_offline(), or open_dead() methods (<type 'exceptions.RuntimeError'>)
Dec 30 16:30:18 ftp weewx[24261]: engine: Unable to load driver: pcapObject must be initialized via open_live(), open_offline(), or open_dead() methods
Dec 30 16:30:18 ftp weewx[24261]:     ****  Exiting...
Dec 30 16:30:18 ftp systemd[1]: weewx.service: Main process exited, code=exited, status=4/NOPERMISSION
Dec 30 16:30:18 ftp systemd[1]: weewx.service: Failed with result 'exit-code'.
Dec 30 16:30:23 ftp systemd[1]: weewx.service: Service hold-off time over, scheduling restart.
Dec 30 16:30:23 ftp systemd[1]: weewx.service: Scheduled restart job, restart counter is at 1.
Dec 30 16:30:23 ftp systemd[1]: Stopped weewx weather system.`

gary-hammer, 2018-12-30:
I am on 0.41 and when I installed 0.43, it fails to load.
See the attached file.
[weewx-interceptor-fail.txt](https://github.com/matthewwall/weewx-interceptor/files/2717536/weewx-interceptor-fail.txt)

matthewwall, 2018-12-30:
please swap lines 328 and 329 so that this:

```
                self.sniffer.setfilter(pcap_filter, 0, 0)
                self.sniffer.open_live(iface, snaplen, pval, timeout_ms)
```

becomes this:

```
                self.sniffer.open_live(iface, snaplen, pval, timeout_ms)
                self.sniffer.setfilter(pcap_filter, 0, 0)
```

gary-hammer, 2018-12-30:
That has it running.
Fails to parse the packets heading towards Ambient though.
See attached file.
[weewx-interceptor-info-1.txt](https://github.com/matthewwall/weewx-interceptor/files/2717547/weewx-interceptor-info-1.txt)

matthewwall, 2018-12-30:
could you run that again, but this time with debug=1 in your weewx configuration file?

matthewwall, 2018-12-30:
that looks like output with debug=0, not debug=1

gary-hammer, 2018-12-30:
Edit debug=1 into weewx.conf
service weewx stop
service weewx start

gary-hammer, 2018-12-30:
Is this not what it should be?
[image]

matthewwall, 2018-12-30:
the `debug=1` must be in the root-level of weewx.conf, not within the `[Interceptor]` stanza.

gary-hammer, 2018-12-30:
You mean right where it says "Set to 1 for extra debug info, otherwise comment it out or set to zero"
Sorry.
See attached
[weewx-interceptor-info-3.txt](https://github.com/matthewwall/weewx-interceptor/files/2717563/weewx-interceptor-info-3.txt)
 file.

matthewwall, 2018-12-30:
please try the attached interceptor

[interceptor-0.43rc1.py.txt](https://github.com/matthewwall/weewx-interceptor/files/2717574/interceptor-0.43rc1.py.txt)

gary-hammer, 2018-12-30:
Seems to miss some data.
See attached.
Back after dinner...
[weewx-interceptor-info-4.txt](https://github.com/matthewwall/weewx-interceptor/files/2717583/weewx-interceptor-info-4.txt)

matthewwall, 2018-12-30:
why do you think it is missing data?

there are two different types of packets happening here - some for api.ambientweather.net and some for wunderground.  the ambientweather packets are coming through with stray HTTP headers in them, while the wunderground packets are clean.  the patch i applied fixes the ambientweather packet parsing.  however, the logic for tracking rain (and possibly other values) does not know how to distinguish ambient packets from wunderground packets.  i will have to fix that.

gary-hammer, 2018-12-30:
Thanks for the explanation. Yes, that is what I thought was lost info.

matthewwall, 2019-12-29:
closing this issue for now.  i am porting interceptor to python3, then after that i will refactor so that it captures only one packet type.

closed 2019-12-29 by matthewwall

gary-hammer, 2019-12-30:
Sounds good to me.