matthewwall/weewx-interceptor #35: Does this work with the new Acurite Access?

kevinkey619, 2018-10-31:
Does this work with the new Acurite Access?

matthewwall, 2018-10-31:
it *should* work with any device that sends data to weather underground, including the acurite access.

please let me know how it works for you.

i should probably make a separate `device_type = acurite-access` (that accepts encrypted access uploads) and a `device_type = wunderground` that simply looks for wunderground protocol.

kevinkey619, 2018-10-31:
Ah. Yeah, it doesn't work with my new Acurite Access. Apparently this device encrypts the data it sends out - unlike the old Acu-Link / smartHUB.

matthewwall, 2018-11-01:
you might try `device_type = observer` - the fine offset hardware also sends to weather underground, so that might work.

could you send me some packet captures from the access?  we should be able to capture the wunderground portion of whatever the access is sending out.  i *think* it is possible to capture the encrypted part too (the part that sends to the chaney instruments servers), but that will take a bit more work.

kevinkey619, 2018-11-01:
What does the screen with just a 1 on it mean? And why can't I get past it?

matthewwall, 2018-11-01:
> What does the screen with just a 1 on it mean? And why can't I get past it?

could you give me a little context, please?  what screen?  what operating system?  what are you running?

kevinkey619, 2018-11-02:
Hi Matt,

So I've tried "acurite-bridge" and "observer", but I'm not having any luck getting the packets parsed from the Acurite Access. Here's a sample packet:

23:33:27.900307 IP W550-0A9B74.3293 > a104-75-128-214.deploy.static.akamaitechnologies.com.http: Flags [P.], seq 1:470, ack 1, win 8192, length 469: HTTP: GET /weatherstation/updateweatherstation.php?ID=KCARAMON46&PASSWORD=e52f792d&softwaretype=myAcuRite&dateutc=now&action=updateraw&realtime=1&rtfreq=36&action=updateraw&realtime=1&rtfreq=35&baromin=28.64&humidity=53&tempf=51.6&windspeedmph=0&winddir=338&windgustmph=0&windgustdir=180&dewptf=35.0&dailyrainin=0.00&rainin=0.00 HTTP/1.1

Disregard the comment regarding "1" as it was intended for a different thread for a different repository altogether.

matthewwall, 2018-11-02:
set `debug=1` in your weewx configuration then run weewx and capture the log.  it should provide what we need to understand why the `GET` requests are not being captured.

first we must be sure that you are capturing (or sniffing) properly

then if that is working we must be sure the parsing is working properly

kevinkey619, 2018-11-03:
> Here are some recent log entries:
>
> Nov  2 09:00:48 raspberrypi weewx[6049]: interceptor: ServerThread: sniff:
> timestamp=1541174448.21 pktlen=42 data=24 c8 6e 0a 9b 74 b8 27 eb 6f e8 24
> 08 06 00 01 08 00 06 04 00 02 b8 27 eb 6f e8 24 c0 a8 06 01 24 c8 6e 0a 9b
> 74 c0 a8 06 41
> Nov  2 09:00:53 raspberrypi weewx[6049]: interceptor: ServerThread: sniff:
> timestamp=1541174453.24 pktlen=42 data=24 c8 6e 0a 9b 74 b8 27 eb 6f e8 24
> 08 06 00 01 08 00 06 04 00 01 b8 27 eb 6f e8 24 c0 a8 06 01 00 00 00 00 00
> 00 c0 a8 06 41
> Nov  2 09:00:54 raspberrypi weewx[6049]: interceptor: MainThread: empty
> queue
> Nov  2 09:00:56 raspberrypi weewx[6049]: interceptor: ServerThread: sniff:
> timestamp=1541174456.0 pktlen=42 data=24 c8 6e 0a 9b 74 b8 27 eb 6f e8 24
> 08 06 00 01 08 00 06 04 00 02 b8 27 eb 6f e8 24 c0 a8 06 01 24 c8 6e 0a 9b
> 74 c0 a8 06 41
> Nov  2 09:01:04 raspberrypi weewx[6049]: interceptor: MainThread: empty
> queue
> Nov  2 09:01:07 raspberrypi weewx[6049]: interceptor: ServerThread: sniff:
> timestamp=1541174467.16 pktlen=42 data=24 c8 6e 0a 9b 74 b8 27 eb 6f e8 24
> 08 06 00 01 08 00 06 04 00 02 b8 27 eb 6f e8 24 c0 a8 06 01 24 c8 6e 0a 9b
> 74 c0 a8 06 41
> Nov  2 09:01:14 raspberrypi weewx[6049]: interceptor: MainThread: empty
> queue
> Nov  2 09:01:16 raspberrypi weewx[6049]: interceptor: ServerThread: sniff:
> [... 53 lines cut ...]
> some identifiers that the Access isn't outputting?
>
> Thanks,
>
> Kevin

I'm quite new to Python. Do I just need to edit /bin/user/interceptor.py
accordingly? Do I need to rebuild or anything?

Thanks,

Kevin

kevinkey619, 2018-11-03:
I got it to work!  

For anyone else searching for an answer and reading this, here's what I did:

Connected the Acurite Access via a bridged connection on the Raspberry Pi.

Specified the "Observer" hardware.

In weewx.conf:

Scroll down to the "" section and make sure it looks like this:

[Interceptor]
    # This section is for the network traffic interceptor driver.

    # The driver to use:
    driver = user.interceptor
    device_type = observer
    mode = sniff
    iface = eth0
    port = 80
    pcap_filter = src net 192.168.6.0/24 && dst port 80
    # NOTE: You'll need to change the above pcap_filter IP address range to match that of the one you assigned to the ethernet port on your own Raspberry Pi. 

    # Specify the hardware device to capture.  Options include:
    #   acurite-bridge - acurite internet bridge
    #   observer - fine offset WH2600/HP1000/HP1003, aka 'observer'
    #   lw30x - oregon scientific LW301/LW302
    #   lacrosse-bridge - lacrosse GW1000U/C84612 internet bridge
    [[sensor_map]]
        pressure = barometer
        barometer = barometer
        altimeter = barometer
        outHumidity = humidity_out
        outTemp = temperature_out
        windSpeed = wind_speed
        windGust = wind_gust
        windDir = wind_dir
        dewpoint = dewpoint
        rain = rain_total

closed 2018-11-03 by kevinkey619