matthewwall/weewx-interceptor #38: Fine Offset WH-2650 and clones display error between absolute and relative pressure

icnmfabro, 2018-11-29:
Hi Matt.

I wanted to tell you that the new WH-2650 Fine Offset weather station with the data transmission bridge has the same problem as the WH-2600 with the barometer value, an error recognized by the issue number 28 (https://github.com/ matthewwall / weewx-interceptor / issues / 28).
This is the track for wunderground intercepted by your driver:

  GET /weatherstation/updateweatherstation.php?ID=xxxxx&PASSWORD=xxxxxx&tempf=42.4&humidity=98&dewptf=41.9&windchillf=42.4&winddir=320&windspeedmph=0.00&windgustmph=0.00&rainin=0.00&dailyrainin=0.00&weeklyrainin=0.01&monthlyrainin=0.01&yearlyrainin=0.01&solarradiation=0.00&UV = 0 & indoortempf = 72.5 & indoorhumidity = 63 & baromin = 29.65 & baromin = 28.38 & lowbatt = 0 & dateutc = 2018-11-26% 2023: 44: 38 & softwaretype = WH2650A_V1.2.1 & action = updateraw & realtime = 1 & rtfreq = 5 HTTP / 1.0
        Accept: * / *
        Host: rtupdate.wunderground.com
        Connection: Close

Is it possible for you to directly modify the driver so that it can present the pressure at a height instead of the absolute pressure?
In case that line number I have to change according to the issue 28 since with the next updates on line 955 there is something else ?.
This problem of absolute pressure presents itself to me with the extension "Meteotemplate", in fact the absolute pressure and not the relative pressure appear on the portal.

I am waiting for your answer.

matthewwall, 2018-12-20:
as i understand it, the station reports a value for 'baromin'.  for some firmware versions, that value is pressure adjusted to sea level, for others it is the gauge pressure.

in the case of your posting, there are *two different* values for 'baromin'!  which one should be used, and how should it be mapped?

icnmfabro, 2018-12-26:
matthew, The first baromir with the plot value of 29.65 is the relative atmospheric pressure, ie the value of the correct pressure with the altitude while the second value corresponds to the absolute pressure, ie the pressure on the sea level.
So the first baromir value should be used, as you did with the WH-2600 station.
For the moment I solved this way:

if software_type == 'WH2600GEN_V2.2.5' or software_type == 'WH2650A_V1.2.1':
          self.LABEL_MAP ['baromin'] = 'pressure'
           logdbg ("firmware% s: using baromin as pressure"% software_type)
           else:
            logdbg ("firmware% s: using baromin as barometer"% software_type)

I would be grateful if you would like to make the standard change on the driver.

matthewwall, 2018-12-30:
patch applied at commit a53bde9, released as weewx-interceptor v0.42

closed 2018-12-30 by matthewwall