matthewwall/weewx-interceptor #39: unable to get data from a LW30x station

gdupuis89, 2018-12-01:
I am new with weewx and unable to get data out of my LW30x weather station ;
weewx is installed on a Raspberry pi whiich is connected on my LAN, i have installed the interceptor driver.
The LW30x is also on my LAN.
when looking at the Syslog file, i have the message: " import of driver failed: [Errno 99]"
This is probably an easy point, but does that ring any bell ?
Thanks
gdupuis89

matthewwall, 2018-12-01:
i need more to go on than that.  run weewx directly:

`weewxd /etc/weewx.conf`

(or see the weewx user guide for details on how to run directly instead of as a daemon)

then you will see much more detail about any failures that are causing the 'import failed' message

m

gdupuis89, 2018-12-02:
after running weewxd /etc/weewx/weewx.conf, i got this (debug=1)

Dec  2 11:32:05 raspberrypi weewx[1153]: engine: Initializing engine
Dec  2 11:32:05 raspberrypi weewx[1153]: engine: Loading station type Interceptor (user.interceptor)
Dec  2 11:32:05 raspberrypi weewx[1153]: interceptor: MainThread: driver version is 0.41
Dec  2 11:32:05 raspberrypi weewx[1153]: interceptor: MainThread: device type: lw30x
Dec  2 11:32:05 raspberrypi weewx[1153]: interceptor: MainThread: sensor map: {'inTemp': 'ot.2:*.*', 'outTemp': 'ot.1:*.*'}
Dec  2 11:32:05 raspberrypi weewx[1153]: interceptor: MainThread: mode is listen
Dec  2 11:32:05 raspberrypi weewx[1153]: interceptor: MainThread: listen on 192.168.1.19:80
Dec  2 11:32:05 raspberrypi weewx[1153]: import of driver failed: [Errno 99] Cannot assign requested address (<class 'socket.error'>)
Dec  2 11:32:05 raspberrypi weewx[1153]: engine: Unable to load driver: [Errno 99] Cannot assign requested address
Dec  2 11:32:05 raspberrypi weewx[1153]:     ****  Exiting…

matthewwall, 2018-12-03:
either there is something already listening on port 80, or you are running weewx as a user who does not have permissions to listen on port 80

gdupuis89, 2018-12-03:
i am not even sure if port 80 is correct on weather station

matthewwall, 2018-12-03:
this has nothing to do with the weather station.  if you use the interceptor in listen mode, then it must listen on a port to which you direct traffic from the lw30x (the redirection is typically done by hijacking the dns, but it could also be done by a router).

if the computer on which you run weewx has a web server running, then that web server is probably already listening on port 80, so the interceptor cannot listen to port 80.  in that case you must either shut down the web server, or configure the web server to reverse proxy to a port on which the interceptor is listening.

if nothing is using port 80, then you probably just need to run weewx as root (non-root processes are not allowed to listen on ports less than 1000)

matthewwall, 2018-12-03:
closing this issue since it is not a bug with the weewx-interceptor driver

closed 2018-12-03 by matthewwall