matthewwall/weewx-interceptor #46: sensor_map issues

zaschf, 2019-02-02:
Hello Matthew,
I have an issue with [[sensor_map]] in the weewx.conf file.
As soon as I add [[sensor_map]] all data fails to map. So far I've only tried solarradiation.
I've attached the weewx.conf file and a log file with the two 'time sections' of a partially mapping and non mapping issue shown.
Any idea why it is doing this.
Can you perhaps create an example sensor_map with the data out of the log file?
Regards,
Frits Schouten.

[weewx.conf.txt](https://github.com/matthewwall/weewx-interceptor/files/2824919/weewx.conf.txt)
[weewx.log.txt](https://github.com/matthewwall/weewx-interceptor/files/2824921/weewx.log.txt)

matthewwall, 2019-02-02:
> As soon as I add [[sensor_map]] all data fails to map. So far I've only tried solarradiation.

your sensor_map with one entry replaces the default map.  so you need specify the default sensor_map, plus your addition, like this:

```
[Interceptor]
    ...
    [[sensor_map]]
        pressure = pressure.*.*
        inTemp = temperature_in.*.*
        inHumidity = humidity_in.*.*
        outTemp = temperature.?*.*
        outHumidity = humidity.?*.*
        windSpeed = windspeed.?*.*
        windDir = winddir.?*.*
        rain = rainfall.?*.*
        txBatteryStatus = battery.?*.*
        rxCheckPercent = rssi.?*.*
        radiation = solarradiation
```

btw, what kind of hardware are you using?  the bridge/smarthub does not know about solar radiation, nor are there any acurite sensors for that.  is this an acurite access device?

i suppose i should add solar radiation to the default map, but maybe not until the access firmware settles down a bit more.

m

zaschf, 2019-02-02:
That's really good to know, Matthew.
I'll implement that right away and I'll let you know how I get on.
I'm using the WeatherSleuth from Aercus.
https://scientificsales.co.nz/aercus-instruments-weathersleuth-professional-ip-weather-station-with-real-time-internet-monitoring/
Cheers,
Frits.

matthewwall, 2019-02-02:
you should use station_type=observer, not station_type=acurite-bridge

that already has a sensor_map that is appropriate for the output from the weathersleuth

matthewwall, 2019-02-02:
oops!  should be device_type=observer:

```
[Interceptor]
    ...
    device_type = observer
    ...
```

zaschf, 2019-02-03:
Ah, better update that...

zaschf, 2019-02-03:
It's all go now, Matthew.
Attached the tailend of syslog
Cheers,
Frits.
[fs.txt](https://github.com/matthewwall/weewx-interceptor/files/2825066/fs.txt)

closed 2019-02-14 by matthewwall