matthewwall/weewx-interceptor #48: interceptor not working anymore

Blisk, 2019-04-16:
My weewx stops to work fev days ago and I try to solve that on weewx google group. And they pointed me here, because it is probabbly interceptor. When I run it manually I get nothing.
I have setup for intercepter.
[Interceptor]
    # This section is for the network traffic interceptor driver.
    
    # The driver to use:
    driver = user.interceptor
    
    # Specify the hardware device to capture.  Options include:
    #   acurite-bridge - acurite internet bridge
    #   observer - fine offset WH2600/HP1000/HP1003, aka 'observer'
    #   lw30x - oregon scientific LW301/LW302
    #   lacrosse-bridge - lacrosse GW1000U/C84612 internet bridge
    device_type = observer
    port = 9999

matthewwall, 2019-04-19:
this is for reporting bugs.

please post support questions at the weewx user group at google groups

closed 2019-04-19 by matthewwall

Blisk, 2019-04-19:
I did, they pointed me here after we didn't manage to solve problem.