matthewwall/weewx-interceptor #51: firmware_version  variable placement 126

udbaker, 2019-04-29:
I have loaded the interceptor and configured but I cannot override the firmware version (defaults to 224).  The firmware_version variable has been added to the weewx.conf file under the Interceptor area but it still defaults to the 224 response code.

Thanks for the help!

matthewwall, 2019-04-29:
please post the configuration

udbaker, 2019-04-29:
# WEEWX CONFIGURATION FILE
#
# Copyright (c) 2009-2019 Tom Keffer <tkeffer@gmail.com>
# See the file LICENSE.txt for your rights.

##############################################################################

# This section is for general configuration information.

# Set to 1 for extra debug info, otherwise comment it out or set to zero
debug = 1

# Root directory of the weewx data file hierarchy for this station
WEEWX_ROOT = /

# Whether to log successful operations
log_success = True

# Whether to log unsuccessful operations
log_failure = True

# How long to wait before timing out a socket (FTP, HTTP) connection
socket_timeout = 20

# Do not modify this. It is used when installing and updating weewx.
version = 3.9.1

##############################################################################

#   This section is for information about the station.

[Station]
    
    # Description of the station location
    location = "Heritage North, Wake Forest, NC"
    
    # Latitude and longitude in decimal degrees
    latitude = 35.960
    longitude = -78.499
    
    # Altitude of the station, with unit it is in. This is downloaded from
    # from the station if the hardware supports it.
    altitude = 350, foot
    
    # Set to type of station hardware. There must be a corresponding stanza
    # in this file with a 'driver' parameter indicating the driver to be used.
    station_type = Interceptor
    
    # If you have a website, you may specify an URL
    #station_url = http://www.example.com
    
    # The start of the rain year (1=January; 10=October, etc.). This is
    # downloaded from the station if the hardware supports it.
    rain_year_start = 1
    
    # Start of week (0=Monday, 6=Sunday)
    week_start = 6

##############################################################################

[Interceptor]
    # This section is for the network traffic interceptor driver.
    
    # The driver to use:
    driver = user.interceptor
   
    firmware_version=126
 
    # Specify the hardware device to capture.  Options include:
    #   acurite-bridge - acurite internet bridge, smarthub, or access
[... 517 lines cut ...]
    [[MySQL]]
        driver = weedb.mysql
        # The host where the database is located
        host = localhost
        # The user name for logging in to the host
        user = weewx
        # The password for the user name (quotes guard against parsing errors)
        password = weewx

##############################################################################

#   This section configures the internal weewx engine.

[Engine]
    
    [[Services]]
        # This section specifies the services that should be run. They are
        # grouped by type, and the order of services within each group
        # determines the order in which the services will be run.
        prep_services = weewx.engine.StdTimeSynch
        data_services = ,
        process_services = weewx.engine.StdConvert, weewx.engine.StdCalibrate, weewx.engine.StdQC, weewx.wxservices.StdWXCalculate
        archive_services = weewx.engine.StdArchive
        restful_services = weewx.restx.StdStationRegistry, weewx.restx.StdWunderground, weewx.restx.StdPWSweather, weewx.restx.StdCWOP, weewx.restx.StdWOW, weewx.restx.StdAWEKAS
        report_services = weewx.engine.StdPrint, weewx.engine.StdReport

udbaker, 2019-05-06:
Found the area in the PY Script and the change worked.  Unfortunately, my bridge is stuck in the permanent firmware upgrade loop.

closed 2019-05-06 by udbaker