matthewwall/weewx-interceptor #54: Issue on LW300 station

olpayras, 2019-06-11:
I use a lw300 oregon. I try to use interceptor driver. All is ok on baro, uv, rain. But have troubles on thermal and humidity datas.

I have 4 thermal sensors : ch1 is ext ch2 is my office, ch3 is my garage, ch4 is my swiming pool.

I have a thermal data wich seems to be an average of all the channels. I look for the code, and i dont see any ExtraTemp in it, so I think, the driver is lost with all the channels.

I do a wireshark capture and that I see is that values are correct on the good channels. Here's the capture, so perhaps you can modify code of interceptor driver to send the good datas to the good values in weewx.

Ie : ExtraTemp1 and ExtraHumid1 be get from :

0000   50 4f 53 54 20 2f 75 70 64 61 74 65 20 48 54 54   POST /update HTT
0010   50 2f 31 2e 30 0d 0a 48 6f 73 74 3a 20 67 61 74   P/1.0..Host: gat
0020   65 77 61 79 2e 77 65 61 74 68 65 72 2e 6f 72 65   eway.weather.ore
0030   67 6f 6e 73 63 69 65 6e 74 69 66 69 63 2e 63 6f   gonscientific.co
0040   6d 0d 0a 43 6f 6e 74 65 6e 74 2d 4c 65 6e 67 74   m..Content-Lengt
0050   68 3a 20 37 35 0d 0a 55 73 65 72 2d 41 67 65 6e   h: 75..User-Agen
0060   74 3a 20 42 6c 61 63 6b 42 6f 78 2f 31 2e 30 2e   t: BlackBox/1.0.
0070   30 38 0d 0a 43 6f 6e 6e 65 63 74 69 6f 6e 3a 20   08..Connection: 
0080   63 6c 6f 73 65 0d 0a 0d 0a 6d 61 63 3d 30 30 30   close....mac=000
0090   34 61 33 36 61 30 35 61 62 26 69 64 3d 38 34 26   4a36a05ab&id=84&
00a0   72 69 64 3d 64 34 26 70 77 72 3d 30 26 68 74 72   rid=d4&pwr=0&htr
00b0   3d 30 26 63 7a 3d 31 26 6f 68 3d 35 33 26 74 74   =0&cz=1&**oh=53**&tt
00c0   72 3d 30 26 6f 74 3d 32 30 2e 38 26 63 68 3d 32   r=0&**ot=20.8**&**ch=2**
00d0   26 70 3d 31                                       &p=1

My weewx with lw300 driver is here : http://www.payrastre.fr/weewx/belchertown/
My weewx with wmr200 using same sensor than lw300 is here, https://meteo.payrastre.fr
so, we can quickly compare.

Here's attached a [wireshark](https://github.com/matthewwall/weewx-interceptor/files/3275731/lw300.zip) capture

matthewwall, 2019-06-12:
you need to define a sensor_map to distinguish the sensors.  in your weewx configuration file, do something like this:

```
[Interceptor]
    ...
    [[sensor_map]]
        pressure = baro.*.*
        outTemp = ot.1:*.*
        outHumidity = oh.1:*.*
        windSpeed = ws.?:*.*
        windGust = wg.?:*.*
        windDir = wd.?:*.*
        rainRate = rr.?:*.*
        rain = rain.?:*.*
        UV = uvh.?:*.*
        extraTemp1 = ot.2.*.*
        extraHumid1 = oh.2.*.*
```

olpayras, 2019-06-12:
Yes, thanks a lot. It's all good now.

closed 2019-06-12 by olpayras