matthewwall/weewx-interceptor #55: Ecowitt HP2550 with PM2.5 and soilmoisture

pchaves1, 2019-06-16:
Hello,

i have a weather station Ecowitt HP2550 with PM2.5 and soilmoisture sensors:
I'm having 3 problems so far with firmware EasyWeatherV1.3.9 and your interceptor:
- RAIN
- PM2.5
- Soilmoisture

Full detailed error:

Jun 16 02:05:24 WeeWX weewx[27143]: interceptor: MainThread: empty queue
Jun 16 02:05:34 WeeWX weewx[27143]: interceptor: MainThread: empty queue
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: ServerThread: GET: ID=i&PASSWORD=XXXX&indoortempf=71.6&tempf=55.2&dewptf=51.6&windchillf=55.2&indoorhumidity=64&humidity=88&windspeedmph=2.0&windgustmph=2.2&winddir=25&absbaromin=29.729&baromin=29.729&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.091&yearlyrainin=0.091&solarradiation=0.00&UV=0&soilmoisture=52&AqPM2.5=309.0&dateutc=2019-06-16%2001:05:39&softwaretype=EasyWeatherV1.3.9&action=updateraw&realtime=1&rtfreq=5
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: raw data: ID=i&PASSWORD=i&indoortempf=71.6&tempf=55.2&dewptf=51.6&windchillf=55.2&indoorhumidity=64&humidity=88&windspeedmph=2.0&windgustmph=2.2&winddir=25&absbaromin=29.729&baromin=29.729&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.091&yearlyrainin=0.091&solarradiation=0.00&UV=0&soilmoisture=52&AqPM2.5=309.0&dateutc=2019-06-16%2001:05:39&softwaretype=EasyWeatherV1.3.9&action=updateraw&realtime=1&rtfreq=5
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: using rain_total 0.091 from yearlyrainin
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: firmware EasyWeatherV1.3.9: baromin is barometer
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter rainin=0.000
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: unrecognized parameter soilmoisture=52
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter monthlyrainin=0.091
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: unrecognized parameter AqPM2.5=309.0
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter realtime=1
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter rtfreq=5
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter PASSWORD=X
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter ID=i
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter softwaretype=EasyWeatherV1.3.9
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter action=updateraw
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: ignored parameter weeklyrainin=0.000
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: raw packet: {'wind_speed': 2.0, 'humidity_in': 64.0, 'temperature_in': 71.6, 'barometer': 29.729, 'windchill': 55.2, 'dewpoint': 51.6, 'humidity_out': 88.0, 'uv': 0.0, 'radiation': 0.0, 'rain': 0.0, 'dateTime': 1560647139, 'pressure': 29.729, 'temperature_out': 55.2, 'wind_dir': 25.0, 'rain_total': 0.091, 'usUnits': 1, 'wind_gust': 2.2}
Jun 16 02:05:43 WeeWX weewx[27143]: interceptor: MainThread: mapped packet: {'barometer': 29.729, 'windchill': 55.2, 'dewpoint': 51.6, 'pressure': 29.729, 'outHumidity': 88.0, 'UV': 0.0, 'radiation': 0.0, 'rain': 0.0, 'dateTime': 1560647139, 'windDir': 25.0, 'outTemp': 55.2, 'windSpeed': 2.0, 'inHumidity': 64.0, 'inTemp': 71.6, 'windGust': 2.2, 'usUnits': 1}
Jun 16 02:05:43 WeeWX weewx[27143]: manager: Added record 2019-06-16 02:05:00 WEST (1560647100) to database 'weewx.sdb'
Jun 16 02:05:43 WeeWX weewx[27143]: manager: Added record 2019-06-16 02:05:00 WEST (1560647100) to daily summary in 'weewx.sdb'

Can you help me fix this problem?

matthewwall, 2019-06-16:
please try version 0.46

closed 2019-06-16 by matthewwall

pchaves1, 2019-06-16:
Many thanks 😊