matthewwall/weewx-interceptor #57: Acurite Access = WeeWX Interceptor

ZXant, 2019-07-24:
Matthew

As always, too many hours already trying to make WeeWX on RasperryPi to connect to Acurite Access, at no result (1st try normal installation; 2nd try driver standalone; etc).

Acurite Access has a Static DHCP through DD-WRT router. The following command, returns no result:

PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/interceptor.py --device=acurite-bridge --mode=sniff --iface=eth0 --filter="src 192.168.1.135 and dst port 80"

Understand there might be other 'strong' approaches (DNS hijack, direct eth0 connection, etc). Please, advice whats "next" and usually works for WeeWX Interceptor and Acurite Access.

Thanks, ZX

jrhelbert, 2019-07-24:
I'm pretty sure the interceptor driver isn't compatible with the Access. Starting with the Access, Acurite sends all the data to their servers using https (or some other form of encrypting protocol) that by design you cant sniff or intercept.

ZXant, 2019-07-25:
Hmmmm... interesting. Thanks Jrhelbert

After much plus hours, I was able to DNS hijack "atlasapi.myacurite.com" (this is the "new" API/myacurite.com address) to RaspberryPi/WeeWX IP address through DD-WRT DNSmasq/DNS Forwarding... still, no luck and stuck. 

Is this a losing battle as Jrhelbert said ("some form of encrypting protocol to their servers")?

If so... should forget about Acurite Access and get any display device, with USB port, that may recognize Atlas? If so, advice/suggestions?

Interesting enough, WeeWX hardware compatibility page does list Acurite Atlas/Access as compatible.

[image]

Thanks, ZXant

matthewwall, 2019-07-25:
weewx does work with the access, but the access must send data in wu format.  configure the access to send wu data.  ideally you would configure it to send to the computer running weewx (use dns hijacking for this).  then configure interceptor to listen (not sniff), and it will simply receive the wu data.

if you prefer to send to wu, then you must configure the interceptor to sniff the traffic (not listen).

please post on the weewx user group for further help on this.

i am sorry there is not yet a guide specifically for intercepter and access

closed 2019-07-25 by matthewwall

ZXant, 2019-07-25:
Thanks Matthew for your time and prompt response. Will try.