matthewwall/weewx-interceptor #62: fineoffset-bridge, WH31 Multi-channel Temp/Humidity Sensors

izzy84075, 2020-01-11:
I just got in my GW1000 and associated sensors. The GW1000 and the WH32 Outdoor Temp/Humidity sensor are working great, but the WH31 Multi-channel Temp/Humidity sensors are not being reported. Looking in the log, I see the following messages, which seem related.

> Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter temp3f=66.56
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter batt1=0
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter batt2=0
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter batt3=0
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter batt4=0
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter humidity4=41
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter humidity1=46
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter humidity2=43
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter humidity3=45
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter temp4f=65.30
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter temp2f=67.10
Jan 11 12:32:49 scd weewx[29314]: interceptor: MainThread: unrecognized parameter temp1f=64.94

I haven't dug too much into the code on this project, but it does look like some of the other stations support multiple remote sensors. Would it be possible to add support for these sensors? I only have four of these sensors, but there's 8 channels, which seem to directly relate to the numbering in the parameter names.

matthewwall, 2020-01-12:
fixed at commit 318b39a

closed 2020-01-12 by matthewwall

izzy84075, 2020-01-12:
This only seems to partially fix this. temp2 and temp3 are working, but temp4 is not, and temp1 still throws errors in the log.

>Jan 12 07:15:18 scd weewx[5288]: interceptor: MainThread: unrecognized parameter batt1=0
Jan 12 07:15:18 scd weewx[5288]: interceptor: MainThread: unrecognized parameter humidity1=39
Jan 12 07:15:18 scd weewx[5288]: interceptor: MainThread: unrecognized parameter temp1f=68.00

Temperature1 /does/ show as a label on one of the report graphs, but nowhere else.

Oddly, temp4 does not throw errors, even though it doesn't show up in the reports.

izzy84075, 2020-01-12:
Also, none of the additional humidity readings seem to show anywhere?

matthewwall, 2020-01-12:
temp1f fixed at commit 930ad3e

if you are not seeing humidity, then that probably means that you are not using ecowitt client mode.  but i added additional humidity to the wuclient as well.

if you want temp4 and others to appear in your weewx plots, then you must modify your schema to capture them.  only extraTemp1-extraTemp3 are in the default schema.  see the weewx customization guide for details.

matthewwall, 2020-01-12:
added pseudo-extensions to wview schema to the default sensor_map for both wuclient and ecowittclient.  that should reduce the amount of tweaking you have to do.

izzy84075, 2020-01-13:
Works perfect, now, thanks! I'm actually only using weewx to get the GW1000 hooked to MQTT, so you adding the defaults in 418aa7c makes it so I don't have to do any other tweaks to WeeWX. ;)

And I have it set to fineoffset-bridge , which I see is deprecated now, but seems to still be equivalent to ecowitt-client mode? I'll go ahead and change it to ecowitt-client, though.