matthewwall/weewx-interceptor #63: Data for new sensor being collected but not stored in db

edsai, 2020-01-16:
I've got an Ecowitt GW1000 and I'm trying to get soil moisture sensors to store data which I then pull out with weewx-influx to store in influxdb and view via Grafana.

Here's the raw data coming back from running weewx in debug:
>>PASSKEY=XXXX&stationtype=GW1000B_V1.5.5&dateutc=2020-01- 16+03:58:02&tempinf=81.7&humidityin=46&baromrelin=29.034&baromabsin=29.034&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.000&yearlyrainin=0.000&totalrainin=0.000&temp1f=69.62&humidity1=79&soilmoisture1=46&pm25_ch1=5.0&pm25_avg_24h_ch1=5.4&wh40batt=1.6&batt1=0&soilbatt1=1.7&pm25batt1=4&freq=915M&model=GW1000_Pro
raw packet: {'humidity_in': 46.0, 'temperature_in': 81.7, 'temperature_1': 69.62, 'humidity_1': 79.0, 'pressure': 29.034, 'rain_rate': 0.0, 'battery_1': 0.0, 'dateTime': 1579147082, 'pm25_ch1': 5.0, 'rain': None, 'soil_moisture_1': 46.0, 'battery_pm25_1': 4.0, 'rain_total': 0.0, 'usUnits': 1}
mapped packet: {'rain': None, 'dateTime': 1579147082, 'pressure': 29.034, 'soilMoist1': 46.0, 'inHumidity': 46.0, 'inTemp': 81.7, 'extraHumid1': 79.0, 'rainRate': 0.0, 'usUnits': 1, 'extraTemp1': 69.62}

I'm not sure if I'm headed down the right path but I added 'soilmoisture1': 'soil_moisture_1' to LABEL_MAP array for EcowittClient class in interceptor.py and then added soilMoist1 = soil_moisture_1 under the Interceptor sensor_map config section in weewx.conf. It's not being stored in the sqlite db or being sent on to influxdb. I do have four tables archive_day_soilMoist1 (through 4) but they're all empty.

I added in pm25 = pm25_ch1 and pm25bat = battery_pm25_1 to weewx.conf to get it to be picked up and it worked. Obviously I'm missing something somewhere. Any help would be greatly appreciated.

matthewwall, 2020-01-16:
please try interceptor 0.52 (commit 8a444d7)

i added labels for soil moisture and their associated batteries

also changed name of pm25 so it will work out-of-the-box with the default sensor_map

anyone with multiple pm25 sensors will have to use sensor_map_extensions to specify the additional pm25 sensors

closed 2020-01-16 by matthewwall

edsai, 2020-01-16:
That worked! Thanks. Can you add another 4 sensors? The GW1000 supports up to 8.

matthewwall, 2020-01-16:
8 sensors as of commit f46b75d

edsai, 2020-01-17:
Awesome. Thanks again. One thing to note, after I updated, I'm no longer get the pm25 readings sent to influxdb. I'm wondering if I'm screwing things up with sensor_map or something else. I didn't modify interceptor after you pushed through the last commits so it's running stock. Below is what I had when pm25 was working but moisture sensors weren't. When I updated to 8a444d7, pm25 stopped being sent over to influxdb. I put sensor_map in so I could assign temperature_1 and humidity_1 to the outside temp and humidity and I read somewhere that once you declare a sensor_map, you need to put all of the mappings in which obviously isn't true because soil moisture is now working fine without it.

[Interceptor]
    # This section is for the network traffic interceptor driver.

    # The driver to use:
    driver = user.interceptor

    # Specify the hardware device to capture.  Options include:
    #   acurite-bridge - acurite internet bridge, smarthub, or access
    #   observer - fine offset WH2600/HP1000/HP1003, ambient WS2902
    #   lw30x - oregon scientific LW301/LW302
    #   lacrosse-bridge - lacrosse GW1000U/C84612 internet bridge
    #   ecowitt-client - any hardware that uses the ecowitt protocol
    #   wu-client - any hardware that uses the weather underground protocol
    #device_type = fineoffset-bridge
    device_type = ecowitt-client
    port = 8000
    [[sensor_map]]
        pressure = pressure.*.*
        inTemp = temperature_in.*.*
        inHumidity = humidity_in.*.*
        outTemp = temperature_1
        outHumidity = humidity_1
        windSpeed = windspeed.?*.*
        windDir = winddir.?*.*
        rain = rainfall.?*.*
        txBatteryStatus = battery.?*.*
        rxCheckPercent = rssi.?*.*
        pm25 = pm25_ch1
        pm25bat = battery_pm25_1
        soilMoist1 = soil_moisture_1

matthewwall, 2020-01-17:
do not specify a sensor_map, that way you'll get the default one.  if you need to change any mappings or add new ones, use [[sensor_map_extensions]]

so you probably want this:
```
[Interceptor]
    ...
    [[sensor_map_extensions]]
        pm25 = pm2_5
        pm25bat = pm25_battery
```

edsai, 2020-01-17:
That worked as well. Thanks for all the help. I really appreciate it. Is there any method of sending money for beer or coffee to you?

matthewwall, 2020-01-17:
just be kind.  or donate to a charity or contribute to an open source project of your choice.

edsai, 2020-06-19:
Hey @matthewwall, I added 4 more soil sensors to bring the total to 8 a couple months ago but the data isn't being stored. I got a few minutes to look into it and it looks like readings aren't being mapped in.

Here's the output of interceptor running in debug:
raw data: PASSKEY=XXXX&stationtype=GW1000B_V1.6.0&dateutc=2020-06-19+20:29:46&tempinf=76.8&humidityin=46&baromrelin=28.860&baromabsin=28.860&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.055&yearlyrainin=14.232&totalrainin=14.232&temp1f=94.64&humidity1=48&soilmoisture1=21&soilmoisture2=30&soilmoisture3=36&soilmoisture4=38&soilmoisture5=52&soilmoisture6=34&soilmoisture7=55&soilmoisture8=28&pm25_ch1=7.0&pm25_avg_24h_ch1=6.1&wh40batt=1.6&batt1=0&soilbatt1=1.6&soilbatt2=1.7&soilbatt3=1.6&soilbatt4=1.7&soilbatt5=1.6&soilbatt6=1.7&soilbatt7=1.5&soilbatt8=1.7&pm25batt1=4&freq=915M&model=GW1000_Pro
raw packet: {'temperature_1': 94.64, 'dateTime': 1592598586, 'humidity_1': 48.0, 'battery_1': 0.0, 'rain_total': 14.232, 'pm2_5': 7.0, 'soil_battery_1': 1.6, 'soil_battery_2': 1.7, 'soil_battery_3': 1.6, 'soil_battery_4': 1.7, 'soil_battery_5': 1.6, 'soil_battery_6': 1.7, 'soil_battery_7': 1.5, 'soil_battery_8': 1.7, 'rain_rate': 0.0, 'rain': 0.0, 'pressure': 28.86, 'pm25_battery': 4.0, 'usUnits': 1, 'humidity_in': 46.0, 'temperature_in': 76.8, 'wh40_battery': 1.6, 'soil_moisture_8': 28.0, 'soil_moisture_1': 21.0, 'soil_moisture_2': 30.0, 'soil_moisture_3': 36.0, 'soil_moisture_4': 38.0, 'soil_moisture_5': 52.0, 'soil_moisture_6': 34.0, 'soil_moisture_7': 55.0}
mapped packet: {'pm2_5': 7.0, 'soilMoist2': 30.0, 'soilMoist4': 38.0, 'rain': 0.0, 'dateTime': 1592598586, 'soilMoist3': 36.0, 'pressure': 28.86, 'soilMoist1': 21.0, 'inHumidity': 46.0, 'inTemp': 76.8, 'extraHumid1': 48.0, 'rainRate': 0.0, 'usUnits': 1, 'extraTemp1': 94.64}

Did I miss a step somewhere?

edsai, 2020-07-19:
Any ideas?