matthewwall/weewx-interceptor #69: Ecowitt: unrecognized parameters to add

git-olicat, 2020-06-02:
Hi!

Some unrecognized parameters could be added (at least for ignore):

lightning_time=[unixtime] - time of last lightning
lightning_num=[int] - count of lightnings
lightning=[int] - distance of last lightning
leak_chN=[0,1] - water sensor N=1..4 (0=ok, 1=alarm)
wh57batt=[int] - battery level 0..5 (max=5) of lightning sensor WH57
leakbattN=[int] - battery level 0..5 (max=5) of water sensor WH55 N=1..4

Regards, Oliver

git-olicat, 2020-06-06:
quick hack:
insert following lines in LABEL_MAP of class EcowittClient:
            'lightning': 'lightning',
            'lightning_time': 'lightning_time',
            'lightning_num': 'lightning_num',
            'wh57batt': 'wh57_battery',
            'leak_ch1': 'leak_1',
            'leak_ch2': 'leak_2',
            'leak_ch3': 'leak_3',
            'leak_ch4': 'leak_4',
            'leakbatt1': 'leak_battery_1',
            'leakbatt2': 'leak_battery_2',
            'leakbatt3': 'leak_battery_3',
            'leakbatt4': 'leak_battery_4',

Additionally you have to adjust the parse-function:

                        # pkt[self.LABEL_MAP[n]] = self.decode_float(data[n]) if data[n] != '' else ''
                        # Oliver, 06.06.20 lightning_time and lightning could be empty - prevent ValueError
                        pkt[self.LABEL_MAP[n]] = self.decode_float(data[n]) if data[n] != '' else ''

Oliver

git-olicat, 2020-06-18:
Because the further processing needs a num it is not the best idea to set the value to an empty string ''.
Perhaps this is a better attempt for the parse-function:

`
                        # pkt[self.LABEL_MAP[n]] = self.decode_float(data[n]) if data[n] != '' else ''
                        # Oliver, 06.06.20 lightning_time and lightning_km could be empty
                        #pkt[self.LABEL_MAP[n]] = self.decode_float(data[n]) if data[n] != '' else ''
                        # Oliver, 18.06.20 further processing requires a num so set 0 to prevent float-conversion error
                        pkt[self.LABEL_MAP[n]] = self.decode_float(data[n]) if data[n] != '' else 0
`
Oliver

icnmfabro, 2020-06-26:
how can I add in the skin template the information of lightning strikes, type counter and distances. also is it possible to insert a chart?

Zaschii, 2020-11-21:
I forward the data with mqtt. But there are no lightning Informations in the mqtt/json string.
Are any changes in weewx.conf necessary? BR

fra202088, 2021-11-07:
#99 
This adds support for WH57 Sebsor and values are consistent with Ecowitt dashboard