matthewwall/weewx-interceptor #70: Import of driver failed: in method 'pcapObject_open_live', argument 2 of type 'char *' (<type 'exceptions.TypeError'>)

ear9mrn, 2020-06-23:
I received the following error when starting weewx...

`ERROR weewx.engine: Import of driver failed: in method 'pcapObject_open_live', argument 2 of type 'char *' (<type 'exceptions.TypeError'>)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 103, in setupStation
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/interceptor.py", line 315, in loader
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      return InterceptorDriver(**config_dict[DRIVER_NAME])
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/interceptor.py", line 2522, in __init__
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      self._device = self.DEVICE_TYPES.get(self._device_type)(**stn_dict)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/interceptor.py", line 1285, in __init__
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      Observer.Parser(), handler=Observer.Handler, **stn_dict)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/interceptor.py", line 427, in __init__
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      iface, pcap_filter, promiscuous)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/interceptor.py", line 469, in __init__
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      self.sniffer.open_live(iface, snaplen, pval, timeout_ms)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****    File "/usr/lib/python2.7/dist-packages/pcap.py", line 108, in open_live
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****      def open_live(self, *args): return _pcap.pcapObject_open_live(self, *args)
Jun 23 09:11:37 time3 weewx[14367] CRITICAL weewx.engine:     ****  TypeError: in method 'pcapObject_open_live', argument 2 of type 'char *'
Jun 23 09:11:37 time3 weewx[14367] CRITICAL __main__: Unable to load driver: in method 'pcapObject_open_live', argument 2 of type 'char *'
Jun 23 09:11:37 time3 weewx[14367] CRITICAL __main__:     ****  Exiting..`

To fix this I edited my weewx.conf Intercept section from this...

```
driver = user.interceptor
device_type = observer
mode = sniff
iface = eth0
pcap_filter = src 192.168.1.22 and dst port 80
```

To this:
```
driver = user.interceptor
address = 192.168.1.22
device_type = observer
mode = sniff
iface = eth0
pcap_filter = dst port 80
```

Looks like the code has changed but the docs have not been updated to reflect the addition of the address parameter and reduced filter description.

Suggest the docs are updated...

Pete.

etamtlosz, 2020-06-23:
Having a similar issue, but your tip didn't help. :(

ear9mrn, 2020-07-04:
try this....

in.. `/usr/share/weewx/user/interceptor.py` around line 460 add the following.

`iface = str(iface)`

etamtlosz, 2020-07-05:
I enden uninstalling everything related to python 3 and installed 2.7, and it works.

drder66, 2020-07-19:
I'm having a lot of issues with python3. Turn on debug and it will immediately go over with a type error. Will start writing up on another thread

stefan1983, 2020-09-22:
What did work for me was using pypcap instead of libpcap. So uninstall the recommended "python-libcacp" and install "python-pypcap" instead.

holyspidoo, 2021-02-13:
After an apt-upgrade all the things, weewx stopped working with this error. Solution was indeed installing "python-pypcap" 

Did not change iface = str(iface). I did change src 192.168.1.22 and dst port 80 to add the address field as well, but I still had same error afterwards, python-pypcap was the fix.