matthewwall/weewx-interceptor #74: No result from Frogitt WH3000SE

drder66, 2020-07-28:
This is another FineOffset clone. The base station WiFi is up and running and sending packets to the ecowitt server. 
Weewx and the interceptor are running on a server which also acts as a firewall and WiFi AP. Hence the capture mode is sniff and the iface is br0. 
Faced with log file full of "empty queue" and "unrecognized packet header"
I've tried running from the command line:
PYTHONPATH=/usr/share/weewx python2 /usr/share/weewx/user/interceptor.py --mode=sniff --address=192.168.10.125 --port=80 --iface=br0 --debug --device=ecowitt-client
But only get the same result. I've tired hacking my way through the code; as far as I can see the bulk of the data comes in on packets ~645 bytes which have POST in them, but they don't seem to be being parsed.

drder66, 2020-07-28:
Added a pcap file with some traffic from the station
[packets.pcap.zip](https://github.com/matthewwall/weewx-interceptor/files/4991463/packets.pcap.zip)

drder66, 2020-07-28:
Other folks seem to have got this working [https://www.wxforum.net/index.php?topic=39744.0](url) and I've followed everything (at least as far as I can see) 
Any ideas what to try/look for next?

gus70, 2020-07-30:
Why not use the WS View app and configure your console to send data to WeeWX system using its "Customized" server upload feature? Much simpler. I recommend sending in Ecowitt format and then set Interceptor to device_type = ecowitt-client