matthewwall/weewx-interceptor #76: Cannot get this to sniff anymore

sgrayban, 2020-08-01:
I upgraded my pi from stretch to buster today and I have been trying to get the extension to work but so far nothing has.

All I get is
`weewx[4562] DEBUG user.interceptor: empty queue`

But if I debug and see if data is getting sent on the wlan0 it does show..
```
# tcpflow -C -0 -i wlan0 -s tcp dst port 80
reportfilename: ./report.xml
tcpflow: listening on wlan0
GET /endpoint?stationtype=AMBWeatherV4.1.7&PASSKEY=XXXXXXXXXXXXXXX&dateutc=2020-08-01+04:58:13&tempinf=77.9&humidityin=43&baromrelin=29.289&baromabsin=27.907&tempf=83.7&humidity=33&winddir=224&windspeedmph=0.0&windgustmph=0.0&maxdailygust=12.5&hourlyrainin=0.000&eventrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.181&totalrainin=29.480&solarradiation=0.00&uv=0 HTTP/1.1
Host: api.ambientweather.net
Connection: close
User-Agent: ESP8266
```

```
# tcpdump -i wlan0
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
22:11:19.088430 ARP, Request who-has 10.3.141.62 tell 10.3.141.1, length 28
22:11:19.292820 ARP, Reply 10.3.141.62 is-at 84:f3:eb:21:92:2d (oui Unknown), length 28
22:12:06.126708 IP 10.3.141.62.4096 > 10.3.141.1.domain: 37909+ A? time.windows.com. (34)
22:12:06.168181 IP 10.3.141.1.domain > 10.3.141.62.4096: 37909 2/0/0 CNAME time.microsoft.akadns.net., A 13.86.101.172 (89)
22:12:06.174481 IP 10.3.141.62.4097 > 13.86.101.172.ntp: NTPv4, Client, length 48
22:12:06.230743 IP 13.86.101.172.ntp > 10.3.141.62.4097: NTPv3, Server, length 48
22:12:14.064571 IP 10.3.141.62.21124 > 172.67.177.12.http: Flags [S], seq 3134441, win 5840, options [mss 1460], length 0
22:12:14.081114 IP 172.67.177.12.http > 10.3.141.62.21124: Flags [S.], seq 3256516366, ack 3134442, win 65535, options [mss 1400], length 0
22:12:14.088640 IP 10.3.141.62.21124 > 172.67.177.12.http: Flags [P.], seq 1:482, ack 1, win 5840, length 481: HTTP: GET /endpoint?stationtype=AMBWeatherV4.1.7&PASSKEY=XXXXXXXXXXXXXXXXXXXXX&dateutc=2020-08-01+05:12:13&tempinf=77.7&humidityin=43&baromrelin=29.292&baromabsin=27.910&tempf=82.0&humidity=33&winddir=224&windspeedmph=0.0&windgustmph=0.0&maxdailygust=12.5&hourlyrainin=0.000&eventrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.181&totalrainin=29.480&solarradiation=0.00&uv=0 HTTP/1.1
```
I am wondering if promiscuous mode isn't working in Pi buster anymore...

sgrayban, 2020-08-01:
Too add to this

`PYTHONPATH=/usr/share/weewx/ python bin/user/interceptor.py --device=observer --mode=sniff --iface=wlan0 --filter="src 10.3.141.62 and dst port 80"`

`mapped packet: {'pressure': 27.927, 'outHumidity': 39.0, 'UV': 0.0, 'radiation': 0.0, 'rain': None, 'dateTime': 1596266654, 'windDir': 209.0, 'outTemp': 76.3, 'windSpeed': 0.2, 'inHumidity': 42.0, 'inTemp': 75.9, 'windGust': 1.1, 'usUnits': 1}`

Running it manually works

closed 2020-08-02 by sgrayban

thiber27, 2020-11-30:
Hello,

I have exactly the same problem, but this issue is closed and no solution for this problem.
Can you give us your solution?
Because i search for several hours and nothing work just "DEBUG user.interceptor: empty queue".

Thank you.