matthewwall/weewx-interceptor #80: Ignored and unrecognized parameters of Ecowitt GW1000

epiniguin, 2020-09-07:
Hi!

There are some ignored and unrecognized parameters that should not be ignored and should be mapped.

wh68batt - unrecognized. Should be recognized and mapped to some database field.
baromrelin -  ignored. Should not be ignored and should be mapped to barometer database field.

Also wh40batt is not ignored but it is not mapped.

`Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: POST: b'PASSKEY=XXXX&stationtype=GW1000A_V1.6.3&dateutc=2020-09-07+12:39:41&tempinf=78.1&humidityin=61&baromrelin=29.991&baromabsin=29.137&tempf=86.7&humidity=35&winddir=33&windspeedmph=5.37&windgustmph=9.17&maxdailygust=11.41&solarradiation=563.93&uv=5&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.551&yearlyrainin=0.551&totalrainin=0.551&wh68batt=1.62&wh40batt=1.6&wh26batt=0&freq=868M&model=GW1000_Pro'
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: raw data: b'PASSKEY=0D07841788B3A48B4D09AD5900E623FD&stationtype=GW1000A_V1.6.3&dateutc=2020-09-07+12:39:41&tempinf=78.1&humidityin=61&baromrelin=29.991&baromabsin=29.137&tempf=86.7&humidity=35&winddir=33&windspeedmph=5.37&windgustmph=9.17&maxdailygust=11.41&solarradiation=563.93&uv=5&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.551&yearlyrainin=0.551&totalrainin=0.551&wh68batt=1.62&wh40batt=1.6&wh26batt=0&freq=868M&model=GW1000_Pro'
Sep  7 15:39:41 dlna weewx[5049] INFO user.interceptor: unrecognized parameter b'PASSKEY=0D07841788B3A48B4D09AD5900E623FD
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter stationtype=GW1000A_V1.6.3
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter baromrelin=29.991
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter maxdailygust=11.41
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter eventrainin=0.000
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter hourlyrainin=0.000
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter dailyrainin=0.000
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter weeklyrainin=0.000
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter monthlyrainin=0.551
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter yearlyrainin=0.551
Sep  7 15:39:41 dlna weewx[5049] INFO user.interceptor: unrecognized parameter wh68batt=1.62
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter freq=868M
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: ignored parameter model=GW1000_Pro'
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: raw packet: {'dateTime': 1599482381, 'usUnits': 1, 'temperature_in': 78.1, 'humidity_in': 61.0, 'pressure': 29.137, 'temperature_out': 86.7, 'humidity_out': 35.0, 'wind_dir': 33.0, 'wind_speed': 5.37, 'wind_gust': 9.17, 'solar_radiation': 563.93, 'uv': 5.0, 'rain_rate': 0.0, 'rain_total': 0.551, 'wh40_battery': 1.6, 'wh26_battery': 0.0, 'rain': 0.0}
Sep  7 15:39:41 dlna weewx[5049] DEBUG user.interceptor: mapped packet: {'dateTime': 1599482381, 'usUnits': 1, 'pressure': 29.137, 'outHumidity': 35.0, 'inHumidity': 61.0, 'outTemp': 86.7, 'inTemp': 78.1,'windSpeed': 5.37, 'windGust': 9.17, 'windDir': 33.0, 'radiation': 563.93, 'rain': 0.0, 'rainRate': 0.0, 'UV': 5.0}
`

guerie74, 2020-09-27:
Hi,

yesterday I managed to setup my EFWS2900 (similar to WS2900) as follows:

- custom server setting in EFWS2900 --> generic-FOSHKplugin --> weewx-interceptor --> weewx

With that setup I got similar complains from the interceptor regarding an ignored parameter **b'humidity=74**. 
The problem is, that there is some garbage **b'** in front of the ignored parameter (similar to your troubles). There is also a garbage tick-character at the end of the string that gets parsed, e.g.:

- **b'** humidity=74&maxdailygust=17.2&weeklyrainin=0.000&humidityin=47&monthlyrainin=2.480&stationtype=EasyWeatherV1.5.3&windspeedmph=2.0&baromabsin=28.240&uv=0&freq=868M&tempf=48.0&dailyrainin=0.000&model=WS2900_V2.01.08&wh65batt=1&totalrainin=36.291&eventrainin=1.823&tempinf=70.7&solarradiation=0.00&PASSKEY=xxx&windgustmph=3.4&baromrelin=29.850&dateutc=2020-09-27+03:59:34&winddir=186&rainratein=0.000&hourlyrainin=0.000 **'**
 
As a small and quick workaround, I changed (added) the following lines (**# RS**) in function _cgi_to_dict
```
def _cgi_to_dict(s):
    if '\'' in s:           # RS
        l=s.split('\'')     # RS
        s=l[1]              # RS
    if '=' in s:
        return dict([y.strip() for y in x.split('=')] for x in s.split('&'))
    return dict()
```

Now I get no more complains in the log and the parameters are displayed correctly in weewx - as far as I see.

guerie74, 2020-09-27:
Hi again,

so I investigated some efforts to dig down the problem and finally ended up with the following... in method do_POST(self), where the payload is read from the http-post message, currently the data is converted with the str() function. Some [research](https://stackoverflow.com/questions/606191/convert-bytes-to-a-string) on **converting bytes to a string** in python showed me, that the conversion via str() is not working properly anymore. Instead of str() the decode("utf-8") function should be used. So in do_POST() the fix looks like:

```
        def do_POST(self):
            # get the payload from an HTTP POST
            length = int(self.headers["Content-Length"])
            #data = str(self.rfile.read(length))                # does not work properly anymore
            data = self.rfile.read(length).decode("utf-8")      # RS
            logdbg('POST: %s' % _obfuscate_passwords(data))
            loginf('POST: length %d, data=[%s] and put...' % (length,_obfuscate_passwords(data)))     # RS just some debug, remove later on
            Consumer.queue.put(data)
            self.reply()

```

Maybe Matthew could have an eye on that and would verify this fix (also forget the workaround from the previous message, it works but it's a dirty workaround).