matthewwall/weewx-interceptor #87: Processing stops - Incorrect Date from Station

adamm82, 2020-11-06:
There seems to be an issue with some units randomly sending a packet with an incorrect date in the future. I have noticed a few posts on some other forums with the same issue. See below logs below:

A correct log (UTC 19:58 - 05:58 local time):
Sep 15 05:59:08 raspberrypi weewx[15243] DEBUG user.interceptor: POST: PASSKEY=XXXX&stationtype=EasyWeatherV1.5.3&dateutc=2020-09-14+19:58:12&tempinf=64.6&humidityin=57&baromrelin=30.095&baromabsin=30.535&tempf=56.3&humidity=99&winddir=29&windspeedmph=1.1&windgustmph=1.1&rainratein=0.000&eventrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.299&totalrainin=82.051&solarradiation=5.14&uv=0&model=WS2900

This is the next log which is incorrect (UTC 05:00 - 15:00 local time) should be 06:00 local:
Sep 15 06:00:11 raspberrypi weewx[15243] DEBUG user.interceptor: raw data: PASSKEY=XXX&stationtype=EasyWeatherV1.5.3&dateutc=2020-09-15+05:00:00&tempinf=64.6&humidityin=57&baromrelin=30.095&baromabsin=30.535&tempf=56.3&humidity=99&winddir=40&windspeedmph=0.7&windgustmph=1.1&rainratein=0.000&eventrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.299&totalrainin=82.051&solarradiation=4.77&uv=0&model=WS2900

From here the time goes back to being correct on following packets, but it is ignored as it it before the last record. Once the time on the incorrect packet passes logging resumes.

It is clearly something with the station itself, but not sure why it is randomly throwing a time in the future, think it is some kind of bug in the latest firmware as I have now seen a couple of other posts with the same issue.

Would a time check be possible? For example if the received record is for more than one hour into the future then discard?

Happy to provide further details if it would assist.

Thanks

etamtlosz, 2021-08-25:
Did you somehow manage to fix this?

etamtlosz, 2021-08-29:
As I understand Matthew doesn't have time to further develop/look into this addon.
The way I'm keeping my records is to send them to influxdb via mqtt. It is overcomplicated, but at least I don't loose any data, since the bug is not affecting mqtt.

I've tried to search for help through the weewx google group, twice during the last year, but no one was able to help me out.
If you find a way to fix this, please update this thread.

benw2, 2022-03-01:
Yeah I have similar issues here, my weather station occasionally uses the wrong time which stuffs everything up. Today it thinks it is the 29th of February 🙄

As a short term hack I have overridden the decode_datetime function to always evaluate to the current time. This is called by the interceptors (weather underground in my case) to parse and decode the current time - around line 706 

https://github.com/matthewwall/weewx-interceptor/blob/4c4a7fbdb109a4ee49c27d1e249c3e4259294525/bin/user/interceptor.py#L706

```
        def decode_datetime(s):
            return int(time.time() + 0.5)   # Just return the current time
```

etamtlosz, 2022-03-01:
Let's see if it helps with my UTC+12 station.