matthewwall/weewx-interceptor #90: TypeError: ord() expected string of length 1, but int found

kq6up, 2020-12-18:
The installation instructions seem to be needing updating.  If you install pypcap by any other means other than # pip3 install pycap.  At that point I get a little further, but failing with the error above.  Latest Raspberry Pi Debian 10 distro with the latest weewx (4.2.0).  Python 3.7 as reflected in the way that I ran it below.

PYTHONPATH=/usr/share/weewx python3.7 /usr/share/weewx/user/interceptor.py  --device=wu-client --mode=sniff --iface=wlan0 --filter="src 10.3.141.99 and dst port 80"
Exception in thread ServerThread:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/share/weewx/user/interceptor.py", line 434, in run_server
    self._server.run()
  File "/usr/share/weewx/user/interceptor.py", line 490, in run
    self.decode_ip_packet(0, pkt, ts)
  File "/usr/share/weewx/user/interceptor.py", line 517, in decode_ip_packet
    (_timestamp, _pktlen, _fmt_bytes(data)))
  File "/usr/share/weewx/user/interceptor.py", line 337, in _fmt_bytes
    return ' '.join(['%02x' % ord(x) for x in data])
  File "/usr/share/weewx/user/interceptor.py", line 337, in <listcomp>
    return ' '.join(['%02x' % ord(x) for x in data])
TypeError: ord() expected string of length 1, but int found

chrisgilldc, 2022-06-21:
Did you ever find a solution to this? I'm trying to reconfigure to split the streams and run in sniff mode and am getting the same error.

eduff0, 2025-02-22:
I have this same issue and commented on issue #117 which is for the same problem.

eduff0, 2025-03-08:
I was able to implement changes to interceptor.py to resolve issue https://github.com/matthewwall/weewx-interceptor/issues/117 and issue https://github.com/matthewwall/weewx-interceptor/issues/90.

See https://github.com/matthewwall/weewx-interceptor/issues/117 for details.