matthewwall/weewx-interceptor #93: question:  Acurite Access - does it contain a barometer sensor?

mrneutron42, 2021-04-28:
I recently bought an Acurite Atlas.  Yesterday, I learned that the outdoor Atlas sensor unit does not include a barometer sensor.
Acurite puts the barometer sensor inside their indoor display units.
I had been hoping to use the weewx-sdr to catch all the data (including barometer).

I wonder if the Acurite Access unit has a barometer sensor inside?

rct, 2021-04-29:
Yes, the Acurite Access box has the barometer sensor inside. None of their outdoor weather stations have a barometer in them.

rct, 2021-05-02:
I wouldn't purchase an Acurite Access just for pressure.  

With the Access you get 5 minute reports instead of the real time updates you are seeing now with rtl_433.

I haven't seriously looked for a 433mhz barometer in a while.  I've seen a bunch of people that just connect a pressure sensor (bosch IIRC) to an RPI. 

I do have an Access, because I got that cheap upgrade from the old smarthub, so I do get pressure from it.  Interestingly it gives readings that are close enough to what I get from the closest airport.  Where the local sensor wins out is when something the readings are dropping fast.

mrneutron42, 2021-05-03:
How can you import airport barometric pressure readings into weewx?

I did some searches and didn't find any USB barometer or pressure sensors.
I did find several Adafruit PCBs with pressure sensors on them.  But, they have i2c or SPI interfaces, not USB.
https://www.adafruit.com/category/114

rct, 2021-05-04:
> How can you import local airport barometric pressure readings into weewx?

Sorry for the confusing, the above screenshots are from Home Assistant. I'm using weather integrations in Home Assistant to poll the US National Weather Service for the closest airport observations (and forecasts).  I'm also pulling some other weather data from Open Weather Map and Dark Sky (until it goes away).

Home Assistant pushes data to MQTT and InfluxDB. However I'm not currently publishing to WeeWX, I've been using Home Assistant as my primary real time console. I also have rtl_433 publishing directly to MQTT, which is how I get that data to Home Assistant and anything else that wants to consume it.  I thought I saw there may be an MQTT option for weewx, but not sure if that is for consuming data or publishing data.  

Yes, the sensors that I've seen connected to an RPI use I2C or SPI.  I haven't seen a USB interfaced version. I think I picked up one of the Adafruit sensors but never got around to hooking it up to a RPI.  Looks straightforward, but I'm currently getting pressure in 5 minute updates from the Acurite Access, though I'd like to eliminate the dependency on that some day.

mrneutron42, 2021-05-13:
I just put together a barometer that can feed WeeWX, using a BMP280 i2c pressure sensor + ESP-12F dev board + Tasmota firmware + MQTT + MQTTSubscribe add-on for WeeWX!
Getting the MQTTSubscribe set up was one of the hardest parts.  I got crucial advice on the weewx-user Google Group and that me over the hump! 
[image]
[image]

closed 2021-05-16 by mrneutron42