matthewwall/weewx-interceptor #102: Two programs; only one customize screen in "WSView"

Tech500, 2022-01-29:
I use weewx to participate in the CWOP program using your intercept driver for ecowitt-client.

Issue arrises when I run "CumulusMX."  There is only one customize page in "WSView."  Both programs run correctly seperately.

[Configuration screen captures](https://drive.google.com/file/d/16tVXus0wS9S_By-bcviVifPQll-CG1k4/view?usp=sharing)

Is there a way to somehow combine or make changes to enable both "weewx" and "CumulusMx?"

Tried using my computer ipAddress 10.0.0.146 for address = 10.0.0.146 with port 8998 only to get "unable to load driver."

William

matthewwall, 2022-01-29:
you should be able to have the gw1000 send to cumulus, then use weewx-interceptor in sniff mode (instead of listen mode) to capture that traffic.

i know this worked with python2, however, i have not yet verified it in python3

(i am working on an update to interceptor now to roll in all of the outstanding issues and merge requests)

Tech500, 2022-01-29:
The ecowitt-client is a Tycon TP3000WC Proweatherstation; not a gw1000, believe princple is the same.  Network wise both; weewx and CumulusMx are on the same subnet.

What needs to be modified in interceptor code to relay traffic to another destination?

Could you describe port mirror?  Have used port forwarding on my Netgear WNDR4300,, not port mirroring.

matthewwall, 2022-01-29:
> What needs to be modified in interceptor code to relay traffic to another destination?

the interceptor has a Consumer class that contains a server and a parser.  the server is either a sniffer (sniffs network traffic to receive data) or a listener (listens on a network port to receive data).  when the server receives data, it sticks the packets onto a queue.  the parser then pops packets off the queue and processes them into weewx LOOP packets.

to do a relay, add another object, lets call it Relay, with its own queue.  when the server receives data, it would put each packet not only onto the parser's queue, but also onto the Relay queue.  the Relay would run on a separate thread that would pop items off its queue and send them to a destination.  this could use many types of destinations and protocols, but i'd start with host:port and http.

> Could you describe port mirroring? Have used port forwarding on my Netgear WNDR4300; not port mirroring.

normally each port on a switch can see only the traffic to/from the device connected to that switch.  so a computer plugging into port 1 cannot see any traffic to/from port 2 (unless, of course, the computer on port 1 is actually sending/receiving data to the device in port 2).

when you enable port mirroring, you let one port see all of the traffic on one (or more) of the other ports.

https://en.wikipedia.org/wiki/Port_mirroring

KC1ELF, 2022-01-29:
Hi Matt,

   I upgraded to Python3 Raspian Bullseye and now can no longer get 
pypcap to work (won't install) which means I don't have the ability to 
"sniff".  You mention interceptor is not verified in python3, so should 
I stop ripping my hair out and go back to python2?  Thank you!

Kind Regards,
Tom Kavanaugh

customize screen in "WSView" (Issue #102)

or Android

matthewwall, 2022-01-29:
the sniffing is the part i have to fix for python3.  if you want to sniff, use python2, at least for now.  there are (were) two network capture libraries that i could use in python2.  i need to figure out what to use now with python3.