matthewwall/weewx-interceptor #107: ModuleNotFoundError: No module named 'pcap'

ear9mrn, 2022-08-01:
Trying to run on Debian 11 with Python 3.9.2 and get the following error.

`ModuleNotFoundError: No module named 'pcap'`

`apt-cache search pcap | grep python`

`python3-impacket - Python3 module to easily build and dissect network protocols`
`python3-pcapy - Python interface to the libpcap packet capture library (Python 3)`
`python3-dpkt - Python 3 packet creation / parsing module for basic TCP/IP protocols`

So installed python3-pcapy and changed:

`import pcap`

to

`import pcapy`

on line 465 of /usr/share/weewx/user/interceptor.py

and got the following error

` File "/usr/share/weewx/user/interceptor.py", line 469, in
__init__
weewx[2668625] CRITICAL weewx.engine:     ****      self.sniffer = pcap.pcapObject()
weewx[2668625] CRITICAL weewx.engine:     ****  NameError: name 'pcap' is not defined`

I then change the following two lines to match the new module:

` 
self.sniffer = pcapy.pcapObject()
self.sniffer = pcapy.pcap(iface, snaplen, pval)
`

Then got more errors:
`
 File "/usr/share/weewx/user/interceptor.py", line 475, in
__init__
weewx[2670687] CRITICAL weewx.engine:     ****      self.sniffer = pcapy.pcap(iface, snaplen, pval)
weewx[2670687] CRITICAL weewx.engine:     ****  AttributeError: module 'pcapy' has no attribute 'pcap'
weewx[2670687] CRITICAL __main__: Unable to load driver: module 'pcapy' has no attribute 'pcap'
`

Now I am out of my depth. Any suggestions to get this working with python 3.9 and the "new" pcapy module or getting the "old" pcap module installed on this system?

Thanks,

Pete.

ear9mrn, 2022-08-15:
I managed to solve my problem. My weather station sent its data to `rtupdate.wunderground.com` on port 80 (this cannot be changed!) so originally I was using sniff mode filtering on the ip address of my weather station 

'pcap_filter = src 192.168.1.22 and dst port 80'

This stopped working as it appears pypcap is currently not available for python3.9 
Oh, by the way, I set my local DNS to redirect traffic to `rtupdate.wunderground.com` to my local machine running weewx, not sure if that was necessary.

Now my setup is a follows:

Local DNS redirects requests to `rtupdate.wunderground.com`  to my local machine running weewx.
On the machine running weewx I set up a site (nginx) that listens for `rtupdate.wunderground.com` traffic and redirects it to port 3000.

`server {
    listen 80;
    server_name rtupdate.wunderground.com;
    location / {
                proxy_pass http://localhost:3000;
}}'

I then set up interceptor to listen rather than sniff (avoids use of pypcap completely!!!).

`[Interceptor]
    driver = user.interceptor
    device_type = observer
    address = localhost
    port = 3000`

I am sure there is a simpler way but it works for me!

User220419, 2023-05-26:
@ear9mrn, great move. Thank you for sharing. I've converted to listen, seems to make more sense than sniffing packets.

My setup I have configured the weather observer hardware to contact a local IP address to update the weather instead of weatherzone, http://x.x.x.x/weatherstation/updateweatherstation.php. This script then performs a HTTP get with the weather data to weatherzone - basically what the observer would have originally been doing before the PHP script was placed in the middle. I've configured interceptor to listen to 3001 (port 3000 was in use) and added another line in the PHP to request the same URL with the payload to 127.0.0.1:3001. Working great!

My weewx interceptor config:

>  driver = user.interceptor
>  device_type = wu-client   
>  address = localhost
>  port = 3001