matthewwall/weewx-interceptor #117: Debug mode:  TypeError: ord() expected string of length 1, but int found

Mikejz84, 2023-11-15:
Attempting to run interceptor in debug mode:  Get the following error.  

`weewxuser@raspberrypi:/usr/share/weewx $ sudo PYTHONPATH=/usr/share/weewx python user/interceptor.py --device=wu-client --mode=sniff --iface=wlan0 --filter="src 192.168.50.9 and dst port 80"
[sudo] password for weewxuser:
Exception in thread ServerThread:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/share/weewx/user/interceptor.py", line 435, in run_server
    self._server.run()
  File "/usr/share/weewx/user/interceptor.py", line 491, in run
    self.decode_ip_packet(0, pkt, ts)
  File "/usr/share/weewx/user/interceptor.py", line 518, in decode_ip_packet
    (_timestamp, _pktlen, _fmt_bytes(data)))
  File "/usr/share/weewx/user/interceptor.py", line 338, in _fmt_bytes
    return ' '.join(['%02x' % ord(x) for x in data])
  File "/usr/share/weewx/user/interceptor.py", line 338, in <listcomp>
    return ' '.join(['%02x' % ord(x) for x in data])
TypeError: ord() expected string of length 1, but int found
`

eduff0, 2025-02-22:
I also have this same issue, in terms of the error on line 338 of interceptor.py.  I am attempting to sniff an AcuRite Access.

[image]

eduff0, 2025-03-08:
I was able to implement changes to **interceptor.py** to resolve issue #117 and issue #90.

There were a handful of changes needed in interceptor.py avoid a series of TypeError exceptions related to character versus bytes data, for use in "Sniff" mode.  (The code worked great without any modifications in "Listen" mode.)  Full disclaimer that I do not know Python but was able to iteratively debug each of the five lines that were causing exceptions in sniffing packets from my AcuRite Access.  In the end, the reports were properly generating (available in a browser via Apache) and the packets continued on their way to WeatherUndergound where I could see that my Acurite Atlas was online.  I speculate that some other lines, utilized for other device types, might also benefit from some changes but I made no attempt to do so. 

Modified **interceptor.py** file: [interceptor.py.TXT](https://github.com/user-attachments/files/19145766/interceptor.py.TXT)

[image]

Revised line #338:
`return ' '.join(['%02x' % x for x in data])`

Revised line #522:
`if len(data) >= 15 and data[12:14] == b'\x08\x00':`

Revised line #524:
`header_len = data[14] & 0x0f`

Revised line #533:
`if b'GET' in _data:`

Revised line #562:
```
if isinstance(self.data_buffer, bytes):
    try:
        decoded_data = self.data_buffer.decode('utf-8')
    except UnicodeDecodeError:
        decoded_data = self.data_buffer.decode('utf-8', errors='replace')
else:
    decoded_data = self.data_buffer
logdbg("sniff: flush %s" % decoded_data)
data = decoded_data
```