matthewwall/weewx-interceptor #120: v5 weewx python3 documentation updates

vinceskahan, 2024-03-20:
Matthew - here's some documentation tweaks that might help folks on v5 get interceptor running under python3 on a pi....

Install and configure the prerequisites

````
# for a pip installation of weewx
source /home/pi/weewx-venv/bin/activate
pip3 install libpcap
pip3 install pypcap
````

Install the driver and enable it

````
weectl extension install https://github.com/matthewwall/weewx-interceptor/archive/master.zip
weectl station reconfigure --no-prompt --driver=user.interceptor
````

And edit weewx.conf to fill in the config entry to listen....

````
[Interceptor]
    driver = user.interceptor
    device_type = ecowitt-client
    mode = listen
    port = 8000
    iface = wlan0
````

The result is it seems (seems...) to listen ok although I don't have any gear emitting data that it could hear.  At a minimum things seem to run ok and the info above might be worth using when you get around to updating the toplevel readme for v5....

vinceskahan, 2024-03-28:
added dpkg commands for getting the pcap items installed (tested in vagrant deb12)

bill-orange, 2025-11-15:
sudo python3 -m pip install libpcap pypcap --break-system-packages    no longer appears to work.  here's the result:

Collecting libpcap
  Using cached libpcap-1.11.0b25-py3-none-any.whl.metadata (15 kB)
Collecting pypcap
  Using cached pypcap-1.3.0.tar.gz (140 kB)
  Preparing metadata (setup.py) ... done
Collecting setuptools>=80.9.0 (from libpcap)
  Using cached setuptools-80.9.0-py3-none-any.whl.metadata (6.6 kB)
Collecting pkg-about>=1.3.6 (from libpcap)
  Using cached pkg_about-2.0.5-py3-none-any.whl.metadata (8.7 kB)
Requirement already satisfied: packaging>=25.0.0 in /usr/lib/python3/dist-packages (from pkg-about>=1.3.6->libpcap) (25.0)
Using cached libpcap-1.11.0b25-py3-none-any.whl (853 kB)
Using cached pkg_about-2.0.5-py3-none-any.whl (7.5 kB)
Using cached setuptools-80.9.0-py3-none-any.whl (1.2 MB)
Building wheels for collected packages: pypcap
  DEPRECATION: Building 'pypcap' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'pypcap'. Discussion can be found at https://github.com/pypa/pip/issues/6334
  Building wheel for pypcap (setup.py) ... error
  error: subprocess-exited-with-error
  
  Ã— python setup.py bdist_wheel did not run successfully.
  â”‚ exit code: 1
  â•°â”€> [352 lines of output]
      Found pcap headers in /usr/include/pcap.h
      Found libraries in /usr/lib/aarch64-linux-gnu/libpcap.so
      found pcap_get_tstamp_precision function
      found pcap_setdirection
      found pcap_setnonblock
      found pcap_compile_nopcap function
      /usr/lib/python3/dist-packages/setuptools/dist.py:599: SetuptoolsDeprecationWarning: Invalid dash-separated key 'description-file' in 'metadata' (setup.cfg), please use the underscore name 'description_file' instead.
      !!
      
              ********************************************************************************
              Usage of dash-separated 'description-file' will not be supported in future
              versions. Please use the underscore name 'description_file' instead.
      
              By 2026-Mar-03, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        opt = self._enforce_underscore(opt, section)
      /usr/lib/python3/dist-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
        warnings.warn(msg)
      /usr/lib/python3/dist-packages/setuptools/dist.py:599: SetuptoolsDeprecationWarning: Invalid dash-separated key 'description-file' in 'metadata' (setup.cfg), please use the underscore name 'description_file' instead.
      !!
      
              ********************************************************************************
              Usage of dash-separated 'description-file' will not be supported in future
              versions. Please use the underscore name 'description_file' instead.
              (Affected: pypcap).
      
              By 2026-Mar-03, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.
      
              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************
      
      !!
        opt = self._enforce_underscore(opt, section)
      running bdist_wheel
      running build
      running build_ext
      building 'pcap' extension
      creating build/temp.linux-aarch64-cpython-313
      aarch64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -DHAVE_PCAP_TSTAMP_PRECISION=1 -DHAVE_PCAP_SETDIRECTION=1 -DHAVE_PCAP_SETNONBLOCK=1 -DHAVE_PCAP_COMPILE_NOPCAP=1 -I/usr/include -I/usr/include/python3.13 -c pcap.c -o build/temp.linux-aarch64-cpython-313/pcap.o
      pcap.c:829:1: warning: â€˜Py_UNICODEâ€™ is deprecated [-Wdeprecated-declarations]
        829 | static CYTHON_INLINE size_t __Pyx_Py_UNICODE_strlen(const Py_UNICODE *u) {
[... 288 lines cut ...]
      pcap.c:28111:53: error: â€˜PyLongObjectâ€™ {aka â€˜struct _longobjectâ€™} has no member named â€˜ob_digitâ€™
      28111 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
            |                                                     ^~
      pcap.c:28166:53: error: â€˜PyLongObjectâ€™ {aka â€˜struct _longobjectâ€™} has no member named â€˜ob_digitâ€™
      28166 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
            |                                                     ^~
      pcap.c:28252:27: error: too few arguments to function â€˜_PyLong_AsByteArrayâ€™
      28252 |                 int ret = _PyLong_AsByteArray((PyLongObject *)v,
            |                           ^~~~~~~~~~~~~~~~~~~
      /usr/include/python3.13/cpython/longobject.h:111:17: note: declared here
        111 | PyAPI_FUNC(int) _PyLong_AsByteArray(PyLongObject* v,
            |                 ^~~~~~~~~~~~~~~~~~~
[... cut ...]
      28517 |     const digit* digits = ((PyLongObject*)b)->ob_digit;
            |                                             ^~
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pypcap
  Running setup.py clean for pypcap
Failed to build pypcap
ERROR: Failed to build installable wheels for some pyproject.toml based projects (pypcap)
pi@Weewx5W:~ $

vinceskahan, 2025-11-15:
Oh well - when they said deprecated I guess they meant deprecated. It's going to happen someday regardless. Other options would likely include adding a specific version of python to your os and using that, and/or using a VM or docker or venv.