matthewwall/weewx-interceptor #123: Supporting Bresser 7-in-1 WiFi

simpz, 2024-12-31:
This is I guess a bug report but will also tell how to setup the weewx-interceptor for the Bresser WiFi 7-in-1 (I guess may well work for the 5-in-1 too). I guess this maybe later added to the application if the authors want.

This device has a nice feature allowing you to upload to "Other Server" that is a specified server URL using the "WUnderground API" so no packet interception needed. On the downside, it seems like it ONLY does HTTPS (but doesn't check the certs) so we need to front end weewx-interceptor with something like nginx to handle this.

First in the android app for the Bresser go to "Weather server" -> Other Server. I don't know what all is needed here and don't know if the URL really matters for weewx-interceptor (just tried to make it look Wunderground like). The "Station ID" and "Station Key" are just random. Update Interval probably up to the user.

```
URL: weewxsrv.lan:8000/weatherstation/updateweatherstation.php
Station ID: 111
Station Key: 222
Update interval: 1 minute
API type: WUnderground API
```

Generated a self signed cert for this host and added a config for this and the GUI to nginx (on port 8002 all up to you of course). Your "public_html" location will vary, I have this running in a container so my location is quite unique.
```

server {
        listen  8000 ssl;
        listen  [::]:8000 ssl;
        server_name weewxsrv.lan;

	ssl_certificate /etc/ssl/certs/nginx-selfsigned.crt;
	ssl_certificate_key /etc/ssl/private/nginx-selfsigned.key;

        location / {
                proxy_set_header Host $http_host;
                proxy_set_header X-Forwarded-Proto $scheme;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header X-Real-IP $remote_addr;
                proxy_redirect off;
                proxy_read_timeout 120;
                proxy_connect_timeout 10;
		proxy_pass http://127.0.0.1:8001/;
                }
}

server {
        listen  8002;
        listen  [::]:8002;
        server_name weewxsrv.lan;

        location /weewx {
		alias /home/weewx/data/public_html;
	}
}

```
Interceptor config weewx.conf addition is simply:
```
[Interceptor]
    driver = user.interceptor
    device_type = wu-client
    port = 8001
```

This gets further but errors with:

`Dec 31 09:57:14 weewxsrv weewx[1114555]: 2024-12-31 09:57:14 weewxd[2]: ERROR user.interceptor: parse failed 1 for ID=111&PASSWORD=2222&dateutc=now&baromin=29.46&tempf=47.1&humidity=82&winddir=318&windspeedmph=2.4&windgustmph=2.4&rainin=0.000&dailyrainin=0.000&uv=0&dewptf=41.7&indoortempf=62.7&indoorhumidity=66&solarradiation=48.885&softwaretype=vws%20versionxx&action=updateraw&realtime=1&rtfreq2.5: dictionary update sequence element #19 has length 1; 2 is required`

This just looks to me like the device is generating a duff CGI with "&rtfreq2.5". Surely that should be "&rtfreq=25.". So to fix this up I modified this with a string substitution correcting this in "_cgi_to_dict".

```
def _cgi_to_dict(s):
    ns=re.sub(r'rtfreq2.5','rtfreq=2.5', s)
    if '=' in ns:
        return dict([y.strip() for y in x.split('=')] for x in ns.split('&'))
    return dict()
```

There is likely a much cleaner way of doing this for individual device brokenness!

I'm now getting data!

Just a few errors now in the logs:
```
Dec 31 11:38:05 weewxsrv weewx[1123472]: 2024-12-31 11:38:05 weewxd[2]: INFO user.interceptor: unrecognized parameter rainin=0.000
Dec 31 11:38:05 weewxsrv weewx[1123472]: 2024-12-31 11:38:05 weewxd[2]: INFO user.interceptor: unrecognized parameter uv=0
```

So under class "WUClient" and LABEL_MAP, changed:
` 'UV': 'uv',`

to 

` 'uv': 'uv',`

And added "rainin" to IGNORED_LABELS in class "WUClient"

```
 IGNORED_LABELS = [
            'ID', 'PASSWORD', 'dateutc', 'softwaretype',
            'action', 'realtime', 'rtfreq',
            'weather', 'clouds', 'rainin',
etc
```
Now I haven't had any rain yet to check this is correct as not so sure what rainin is for really but it's ignored for other device types, so probably the right thing to do?

This keeps it quiet. 
Might help someone else and allow patches to be applied to main branch if the developers want.

simpz, 2025-06-30:
[Wi-Fi TB 7-in-1 RC Weather Station](https://www.bresseruk.com/p/bresser-wi-fi-tb-7-in-1-rc-weather-station-7002582?gad_source=4&gclid=CjwKCAiA9bq6BhAKEiwAH6bqoKFgu8pVeM3SDo11O8sgaRznQzPgd0EmU78ga7x4r4yy_eEq2uNJnRoCk78QAvD_BwE)

This was the one I had. But  you should see some traffic to your server.
Can you watch the router traffic to see if it does send to Wunderground and then maybe you can use the interceptor mode.

koanya, 2025-06-30:
No, I see no traffic to my server. The upload to ProWeatherLive also doesn't work (but I can download data from ProWeatherLive). I will test with Wunderground but I don't know if my router can show traffic.

simpz, 2025-07-05:
@zzottel thanks for your additional info here. I'm confused why no one else sees the options I do in the app. Here is what I see, my firmware is 3.00 and this is the android

[image]

[image]

[image]

simpz, 2025-07-11:
Just for completeness here is my firmware screen from the app. 

[image]

On mapping rainin to rainRate, if you add this to IGNORED_LABELS as I did,  what the Interceptor WUClient does is uses the "dailyrainin" (in our Weatherstation's case) and compares it to last one it got to work out the rainRate.

This seems preferred from the interceptor by this comment:

```
# different firmware seems to report rain in different ways.
# prefer to get rain total from the yearly count, but if
 # that is not available, get it from the daily count.
```
,  but it maybe just for inconsistency between different weather stations and their implementation of the WU protocol. 
Just to note this in this thread, I don't really know if one approach is better than the other in the Bresser's case.